bblanchon/ArduinoJson #909: Maybe Unitiailized Variable Warning on gcc 8.2

furkanusta, 2019-02-21:
Because of the name of the project I wasn't sure the target platform is supposed to be an Arduino but when I comment out -Werror the library builds fine and passes the tests. That is why, I've opened this issue.

I am on x86_64, Fedora 29. Using g++ 8.2

And these are the warnings I got

> ../test/JsonVariant/subscript.cpp: In function ‘void ____C_A_T_C_H____T_E_S_T____0()’:
> ../test/JsonVariant/subscript.cpp:71:17: warning: ‘var.ArduinoJson::JsonVariant::_content.ArduinoJson::Internals::JsonVariantContent::asArray’ may be used uninitialized in this function [-Wmaybe-uninitialized]
>      JsonVariant var = _jsonBuffer.parseArray("[{}]");

> ../test/JsonVariant/subscript.cpp:67:42: warning: ‘var.ArduinoJson::JsonVariant::_content.ArduinoJson::Internals::JsonVariantContent::asArray’ may be used uninitialized in this function [-Wmaybe-uninitialized]
>      REQUIRE(std::string("world") == var[0]);

> ../test/JsonVariant/subscript.cpp:19:55: warning: ‘var.ArduinoJson::JsonVariant::_content.ArduinoJson::Internals::JsonVariantContent::asArray’ may be used uninitialized in this function [-Wmaybe-uninitialized]
>      REQUIRE(std::string("element at index 0") == var[0]);

bblanchon, 2019-02-22:
Hi @furkanusta,

Yes, despite the name of the library, you can use it on any platform.

I don't manage to reproduce the warning.
I tried on a virtual machine running Ubuntu 18.04.2 with GCC 8.2.0, but I got no error.
I'll try to get a Fedora 29 and keep you informed.
However, it really looks like a false positive.

BTW, ArduinoJson is a header-only library, you don't have to compile it.
Simply download the header file from:
https://github.com/bblanchon/ArduinoJson/releases/download/v5.13.4/ArduinoJson-v5.13.4.h

Regards,
Benoit

bblanchon, 2019-02-22:
I just tested with Fedora 29 with GCC 8.2.1, and I still have no warnings.
Can you help me reproduce this issue?

furkanusta, 2019-02-22:
I've tried again it looks like it only appears in Release mode, not in Debug mode. Maybe an optimization causes a false positive for gcc.

bblanchon, 2019-02-23:
OK, it really looks like a false positive.
It checked ArduinoJson 6, and it builds without any warning.
I don't think it worth fixing version 5.

furkanusta, 2019-02-24:
Yeah, since it seems to be a false positive and 6.x is working there is no need to backport to 5. Thanks for the quick check.

closed 2019-02-24 by furkanusta

bblanchon, 2019-02-24:
You're welcome.
Thank you for reporting this issue.