bblanchon/ArduinoJson #911: Trouble with Json and sketch

dhpete, 2019-02-26:


dhpete, 2019-02-26:
Hey guys, sorry for the Noob question. I've been struggling with this sketch for a week or so, so I reinstalled to the latest IDE and installed the new Json library but still having issues. This is the sketch

extern "C" {
#include "user_interface.h"
}
#include <FS.h>
#include <Font6x8_meins_nodemcu_v.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ESP8266mDNS.h>        // Include the mDNS library
#include <WebSocketsServer.h>
#include <ArduinoJson.h>

#include <WiFiUdp.h>
#include <ArduinoOTA.h>

//===== ↓↓↓ SETTINGS ↓↓↓ =====

char boardname[] = "ledwall";

#define SENSORPIN D5 //USE SOMETHING OTHER THAND D0, 2, 8 - stop ESP from booting
#define BOARDLED 16
#define sensor false

int LEDRefreshInterval = 50;
os_timer_t refreshTimer;
//os_timer_t ntpTimer;
time_t lastTime;
#include "TimeThings.h"
//Timers

//LED
// Params for width and height
const uint8_t kMatrixWidth = 20;
const uint8_t kMatrixHeight = 18;

#define drawColon true
#define secondsBar false

#define NUM_LEDS kMatrixWidth*kMatrixHeight
#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>
#define LED_PIN         1
#define COLOR_ORDER     GRB
#define CHIPSET         WS2812B
CRGB leds_plus_safety_pixel[ NUM_LEDS + 1];
CRGB* const leds( leds_plus_safety_pixel + 1);
// Param for different pixel layouts
const bool    kMatrixSerpentineLayout = true;

int volts = 12;
int milliamps = 10000;

//===== ↑↑↑ SETTINGS ↑↑↑ =====

int brightness = 120;
int manualBrightness = -1;
int wallMode = 0;
long effectCounter = 0;

bool sunrise = false;
int sunriseDuration = 10;
int sunriseBrightness = 255;
int sunriseMinuteOfDay = 1000;
File fsUploadFile;
String tickerString = "";

#include "LEDMatrixThings.h" // bright, drawLetter, darwTime, drawDate 
[... 420 lines cut ...]
           JsonObject& json = jsonBuffer.createObject();

                              ^

myLED_wall:285:16: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

           json.printTo(configFile);

                ^

Multiple libraries were found for "Font6x8_meins_nodemcu_v.h"
 Used: C:\Users\Pete\Documents\Arduino\libraries\Font
 Not used: C:\Program Files (x86)\Arduino\libraries\Font
 Not used: C:\Program Files (x86)\Arduino\libraries\Font
 Not used: C:\Program Files (x86)\Arduino\libraries\Font
 Not used: C:\Program Files (x86)\Arduino\libraries\Font
exit status 1
'DynamicJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

If someone smarter than me could take a look at this and tell me where Im wrong, I would forever be in your debt. Thanks

bblanchon, 2019-02-28:
Hi @dhpete,

I think ArduinoJson 6 is installed on your computer, but the program is written for ArduinoJson 5.
To solve this problem, open the Library Manager and downgrade ArduinoJson to 5.13.4.

Regards,
Benoit

dhpete, 2019-02-28:
Thanks, I'll give it a go.

closed 2019-03-11 by bblanchon