bblanchon/ArduinoJson #912: Compile warning "unused variable" in FloatTraits.hpp

m-andersen, 2019-02-26:
I get the following compile warning under linux

/home/build/toolchains/arm-2009q1/bin/arm-none-linux-gnueabi-g++ -O0 -g3 -Wall -I. -I/home/andersma/work/disccu/cu-app/src/CU/../../../libjson/src/ArduinoJson -c -fmessage-length=0 -O3 -MMD -MP -MF"main.d" -MT"main.o" -o "main.o" "../main.cpp"
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/../Serialization/../TypeTraits/FloatTraits.hpp: In static member function 'static T ArduinoJson::Internals::FloatTraits<T, 8u>::forge(uint32_t, uint32_t) [with T = double]':
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/../Serialization/../TypeTraits/FloatTraits.hpp:56:   instantiated from 'static T ArduinoJson::Internals::FloatTraits<T, 8u>::positiveBinaryPowerOfTen(int) [with T = double]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/../Serialization/FloatParts.hpp:67:   instantiated from 'static int16_t ArduinoJson::Internals::FloatParts<TFloat>::normalize(TFloat&) [with TFloat = double]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/../Serialization/FloatParts.hpp:25:   instantiated from 'ArduinoJson::Internals::FloatParts<TFloat>::FloatParts(TFloat) [with TFloat = double]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:91:   instantiated from 'void ArduinoJson::Internals::JsonWriter<Print>::writeFloat(TFloat) [with TFloat = double, Print = ArduinoJson::Internals::StreamPrintAdapter]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:70:   instantiated from 'static void ArduinoJson::Internals::JsonSerializer<Writer>::serialize(const ArduinoJson::JsonVariant&, Writer&) [with Writer = ArduinoJson::Internals::JsonWriter<ArduinoJson::Internals::StreamPrintAdapter>]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp:35:   instantiated from 'typename ArduinoJson::Internals::EnableIf<(! ArduinoJson::Internals::StringTraits::has_append), unsigned int>::type ArduinoJson::Internals::JsonPrintable<T>::printTo(Print&) const [with Print = ArduinoJson::Internals::StreamPrintAdapter, T = ArduinoJson::JsonVariant]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp:42:   instantiated from 'std::ostream& ArduinoJson::Internals::JsonPrintable<T>::printTo(std::ostream&) const [with T = ArduinoJson::JsonVariant]'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/JsonVariantImpl.hpp:122:   instantiated from here
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/../Serialization/../TypeTraits/FloatTraits.hpp:105: warning: unused variable 'integerBits'
/home/andersma/work/src/CU/../../../libjson/src/ArduinoJson/Deserialization/../Serialization/../Serialization/../TypeTraits/FloatTraits.hpp:105: warning: unused variable 'floatBits'

Compiling using the following compiler:
arm-none-linux-gnueabi-g++ (Sourcery G++ Lite 2009q1-203) 4.3.3
Copyright (C) 2008 Free Software Foundation, Inc.

I have included in my CPP file using
#include "ArduinoJson.hpp"

and then reference it using namespace.
   ArduinoJson::DynamicJsonBuffer m_JsonBuffer;
   ArduinoJson::JsonObject &m_JsonRoot;

bblanchon, 2019-02-27:
Hi @m-andersen,

Thank you very much for reporting this warning.

Newer versions of GCC don't produce this warning, but I managed to reproduce it with GCC 4.4.7:
https://wandbox.org/permlink/B3UIeVtZtmpG6OGg

I'll try to find a workaround, or I'll mute this warning.

Regards,
Benoit

bblanchon, 2019-02-27:
I finally understood why the Continuous Integration doesn't have this warning.
It's because I use `#include <ArduinoJson.h>` in the tests, instead of `#include "ArduinoJson.h"`.
GCC treats the header as a "system header", so it doesn't report all the warnings.

m-andersen, 2019-02-27:
Hi

Nice to hear it can be reproduced. I guess it is easy to fix. Let me know when you have push an update to GitHub and I will pick it up and test it again.

br
Martin

bblanchon, 2019-02-28:
I fixed this warning in branches 5.x and 6.x.

closed 2019-02-28 by bblanchon