bblanchon/ArduinoJson #913: problem looping over multiple arrays with multiple objects.

novski, 2019-02-27:
I have this Json File in /tmp/sensors.json:
```
{
  "current": [
    {
      "label": "current Sensor",
      "objId": 1,
      "type": "text",
      "deviceAddress": 64,
      "deviceName": "INA219",
      "busMilliVolts": 0.004976,
      "shuntMicroVolts": 0.02823,
      "busMicroAmps": 0,
      "busMicroWatts": 1.477
    }
  ],
  "switches": [
    {
      "label": "Heater Switch",
      "function": "switch",
      "objId": 1,
      "type": "state",
      "pin": 86,
      "pinValue": 0
    },
    {
      "label": "Waterlevel",
      "function": "switch",
      "objId": 2,
      "type": "state",
      "pin": 5,
      "pinValue": 1
    }
  ],
  "temperature": [
    {
      "label": "Internal Temperature",
      "objId": 1,
      "type": "text",
      "temperature": 36.1,
      "humidity": 23.5
    },
    {
      "label": "Heater Temperature",
      "objId": 2,
      "type": "text",
      "temperature": 26,
      "humidity": 0
    }
  ]
}
```
And i try to parse it like this: 
```
void readJson(){
  File sensorReadFile = FileSystem.open("/tmp/sensors.json", FILE_READ); 
  DynamicJsonDocument doc(readCapacity);
    deserializeJson(doc, sensorReadFile);
    int memoryUsed = doc.memoryUsage(); 
    Console.print("memory used: ");Console.println(memoryUsed); // = 313
    
    JsonObject obj = doc.as<JsonObject>();
    int obj_length=obj.size();
    Console.print("obj_length: ");Console.println(obj_length); // = 2
    
    for(int i=0; i<obj_length;i++){ 
      Console.print("i ="); Console.println(i); 
        JsonObject repo= doc[i];
        int pin = repo["pin"];
        Console.print("pin:");Console.println(pin); // = pin:0
        if (pin==86){
          Console.print("pinValue86:");
        }
    }
    sensorReadFile.close();
}  
```

Why is .size() only 2? (i see 3 Arrays; current, switches & temperature)

And how can i access **pinValue** of pin:86?

Thanks a lot for help...

platform: Arduino 1.8.8
compiler: avrdude: Version 6.1,
ArduinoJson V6.9.0

bblanchon, 2019-02-28:
Hi @novski,

Please see [JsonObject::begin() / end()](https://arduinojson.org/v6/api/jsonobject/begin_end/?utm_source=github&utm_medium=issues)

Regards,
Benoit

novski, 2019-02-28:
hi, thanks for that. I was now able to print the keys of 2 of my arrays. 
But there are **three** arrays in that file so something is not working correct. 
I used this on a arduino uno to make it a bit shorter and faster to debug:
```
#include <ArduinoJson.h>
const size_t capacity = JSON_ARRAY_SIZE(1) + 2*JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(3) + 2*JSON_OBJECT_SIZE(5) + 2*JSON_OBJECT_SIZE(6) + JSON_OBJECT_SIZE(9);
const char* json = "{\"current\":[{\"label\":\"current Sensor\",\"objId\":1,\"type\":\"text\",\"deviceAddress\":64,\"deviceName\":\"INA219\",\"busMilliVolts\":0.00498,\"shuntMicroVolts\":0.02485,\"busMicroAmps\":0,\"busMicroWatts\":1.239}],\"switches\":[{\"label\":\"Heater Switch\",\"function\":\"switch\",\"objId\":1,\"type\":\"state\",\"pin\":86,\"pinValue\":0},{\"label\":\"Waterlevel\",\"function\":\"switch\",\"objId\":2,\"type\":\"state\",\"pin\":5,\"pinValue\":1}],\"temperature\":[{\"label\":\"Internal Temperature\",\"objId\":1,\"type\":\"text\",\"temperature\":35,\"humidity\":25.5},{\"label\":\"Heater Temperature\",\"objId\":2,\"type\":\"text\",\"temperature\":26,\"humidity\":0}]}";
void setup() {
Serial.begin(9600);
}
void loop() {
readJson();
}
void readJson(){ 
  DynamicJsonDocument doc(capacity);
    deserializeJson(doc, json);
    int memoryUsed = doc.memoryUsage();
    Serial.print("memory used: ");Serial.println(memoryUsed);
    JsonObject root = doc.as<JsonObject>();
    int root_length=root.size();
    Serial.print("obj_length: ");Serial.println(root_length);
    for (JsonPair kv : root) {
      Serial.println(kv.key().c_str());
      Serial.println(kv.value().as<char*>());
      delay(500);
    }
} 
```
i get this:
```
23:28:43.309 -> memory used: 312
23:28:43.343 -> obj_length: 2
23:28:43.343 -> current
23:28:43.343 -> 
23:28:43.524 -> switches
23:28:43.524 -> 
23:28:43.699 -> memory used: 312
23:28:43.734 -> obj_length: 2
23:28:43.734 -> current
23:28:43.770 -> 
23:28:43.910 -> switches
23:28:43.910 -> 
23:28:44.124 -> memory used: 312
23:28:44.124 -> obj_length: 2
23:28:44.159 -> current
23:28:44.159 -> 
23:28:44.329 -> switches
23:28:44.329 -> 
```
howcome it does not find the array with name "temperature" ?

bblanchon, 2019-03-01:
Hi @novski,

[deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/?utm_source=github&utm_medium=issues) returns [NoMemory](https://arduinojson.org/v6/api/misc/deserializationerror/?utm_source=github&utm_medium=issues) because the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/?utm_source=github&utm_medium=issues) is too small.
See demo: https://wandbox.org/permlink/YwOFIB9OZ78cupq4

Indeed, since the input is read-only, we must increase the capacity.
The [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/?utm_source=github&utm_medium=issues) says we must add 410.
See demo: https://wandbox.org/permlink/8lMfWczOgTyc3cZH

Now, we should see the three arrays "current," "switches," and "temperature."
To enumerate the values in the arrays, we must call [JsonArray::begin() / end()](https://arduinojson.org/v6/api/jsonarray/begin_end/?utm_source=github&utm_medium=issues); but before, we must cast the value to a [JsonArray](https://arduinojson.org/v6/api/jsonarray/?utm_source=github&utm_medium=issues) with [JsonVariant::as<JsonArray>()](https://arduinojson.org/v6/api/jsonvariant/as/?utm_source=github&utm_medium=issues).

```c++
for (JsonPair kv : doc.as<JsonObject>()) {
  Serial.print(kv.key().c_str());
  Serial.println(":");
  for (JsonObject obj : kv.value().as<JsonArray>() {
    Serial.print(" - ");
    Serial.println(obj["label"].as<char*>());
  }
}
```

The program above prints:

```
current:
 - current Sensor
switches:
 - Heater Switch
 - Waterlevel
temperature:
 - Internal Temperature
 - Heater Temperature
```

See demo: 
https://wandbox.org/permlink/ViWHYltToOJBGnle

closed 2019-03-11 by bblanchon