bblanchon/ArduinoJson #914: Does not compile on Raspberry Pi 3

umar14, 2019-02-28:
### * Target Platform
```
Raspbian GNU/Linux 9 (stretch)
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
```

### *** Compiler model and version**

- GCC
```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1)
```

- CMAKE:
`cmake version 3.14.0-rc1`

- CMakeLists.txt:

```
# ArduinoJson - arduinojson.org
# Copyright Benoit Blanchon 2014-2019
# MIT License

cmake_minimum_required(VERSION 3.0)
project(ArduinoJson)

enable_testing()

add_definitions(-DARDUINOJSON_DEBUG)
if(CMAKE_CXX_COMPILER_ID MATCHES "(GNU|Clang)")
        add_compile_options(-g -O0)
endif()

if(${COVERAGE})
        set(CMAKE_CXX_FLAGS "-fprofile-arcs -ftest-coverage")
endif()

include_directories(${CMAKE_CURRENT_LIST_DIR}/src)
add_subdirectory(third-party/catch)
add_subdirectory(test)
add_subdirectory(fuzzing)
```

- CMake Output
```
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/cplusplus/ArduinoJson
```

- MakeFile
```
# CMAKE generated file: DO NOT EDIT!
# Generated by "Unix Makefiles" Generator, CMake Version 3.14

# Default target executed when no arguments are given to make.
default_target: all

.PHONY : default_target

# Allow only one "make -f Makefile2" at a time, but pass parallelism.
.NOTPARALLEL:

#=============================================================================
[... 470 lines cut ...]
In file included from /home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/StringAdapters.hpp:7:0,
                 from /home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/SerializedValue.hpp:7,
                 from /home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/VariantData.hpp:7,
                 from /home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:7,
                 from /home/pi/cplusplus/ArduinoJson/src/ArduinoJson.hpp:9,
                 from /home/pi/cplusplus/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/pi/cplusplus/ArduinoJson/test/ElementProxy/add.cpp:5:
/home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp: In instantiation of ‘T* ArduinoJson690_010::MemoryPool::allocRight() [with T = ArduinoJson690_010::VariantSlot]’:
/home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp:50:36:   required from here
/home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp:91:12: error: cast from ‘char*’ to ‘ArduinoJson690_010::VariantSlot*’ increases required alignment of target type [-Werror=cast-align]
     return reinterpret_cast<T*>(allocRight(sizeof(T)));
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp: In instantiation of ‘T* ArduinoJson690_010::MemoryPool::allocRight() [with T = ArduinoJson690_010::StringSlot]’:
/home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp:107:35:   required from here
/home/pi/cplusplus/ArduinoJson/src/ArduinoJson/Array/../Variant/../Misc/../Strings/../Memory/MemoryPool.hpp:91:12: error: cast from ‘char*’ to ‘ArduinoJson690_010::StringSlot*’ increases required alignment of target type [-Werror=cast-align]
cc1plus: all warnings being treated as errors
test/ElementProxy/CMakeFiles/ElementProxyTests.dir/build.make:62: recipe for target 'test/ElementProxy/CMakeFiles/ElementProxyTests.dir/add.cpp.o' failed
make[2]: *** [test/ElementProxy/CMakeFiles/ElementProxyTests.dir/add.cpp.o] Error 1
CMakeFiles/Makefile2:208: recipe for target 'test/ElementProxy/CMakeFiles/ElementProxyTests.dir/all' failed
make[1]: *** [test/ElementProxy/CMakeFiles/ElementProxyTests.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2
```

```

bblanchon, 2019-02-28:
Hi @umar14,

Thank you for reporting this.
I fixed most of the warnings in a local branch.
Hopefully, it'll be fixed tomorrow.

Regards,
Benoit

bblanchon, 2019-03-01:
Hi @umar14,

I fixed the three warnings in the main branch.

BTW, do you know that you don't have to compile the library?
ArduinoJson is a header-only library, so you only need to download ArduinoJson.h from the [release page](https://github.com/bblanchon/ArduinoJson/releases).

Regards,
Benoit

closed 2019-03-01 by bblanchon

umar14, 2019-03-01:
Hello,

Yes I realized that I didn't need to compile it after I opened the issue here. Therefore closed it immediately. Thank you so much for your reply. It's a great library and I have found it to be very useful in my Arduino projects. Wanted to test it on Raspberry Pi as well. Included the Headers in my project and it works flawlessly. Keep up the good work!!

Regards,
Umar