bblanchon/ArduinoJson #915: DynamicJsonBuffer is a class from ArduinoJson 5

probonopd, 2019-02-28:
Code that was still building fine ([build log](https://travis-ci.org/probonopd/WirelessPrinting/builds/498559031), "Using library ArduinoJson at version 5.13.4 in folder: /home/travis/Arduino/libraries/ArduinoJson") a couple of days ago is no longer building now, saying

> ESP8266WirelessPrintAsync:239:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

This is highly annoying. It is a hassle having to change our sketch because ArduinoJson is a moving target.

In your build script, we are explicitly [using](https://github.com/probonopd/WirelessPrinting/blob/GMagician-code-rewamping/.travis.yml#L24) `git clone -o ed98ea4 https://github.com/bblanchon/ArduinoJson` because we don't want a moving target.

Still, we are getting the error message ([build log](https://travis-ci.org/probonopd/WirelessPrinting/builds/498559031), "Using library ArduinoJson at version 6.9.0 in folder: /home/travis/Arduino/libraries/ArduinoJson".

Do you have an idea why this might be happening?

Why are we getting ` library ArduinoJson at version 6.9.0 in folder: /home/travis/Arduino/libraries/ArduinoJson ` when we are doing `git clone -o ed98ea4 https://github.com/bblanchon/ArduinoJson`?

bblanchon, 2019-02-28:
Hi @probonopd,

I changed the default branch to 6.x because it's now the stable release.
If you want ArduinoJson 5, you must specify the branch 5.x:

    git clone -b 5.x ArduinoJson https://github.com/bblanchon/ArduinoJson

Please apologize for the trouble.

Regards,
Benoit

probonopd, 2019-02-28:
Hi @bblanchon thanks for your response. It's great that you are improving the library, even if this means changes for library users. I will definitely use the new version for new projects going forward.

In this case, however, I was assuming that specifying `git clone -o ed98ea4` and thus freezing the library to the exact version I used when I wrote the sketch would guarantee that things would continue to work forever without changes in the sketch, but apparently that is not the case.

probonopd, 2019-02-28:
Thanks, `-b 5.x` worked for me.

closed 2019-02-28 by probonopd