bblanchon/ArduinoJson #916: Feature Request : Partial Serialize Stream

capricorn-one, 2019-03-01:
Running v6.9 on a Teensy 3.,2 via Visual Micro compiler.

Maybe I'm not using the library right, but I have an array of 2k floats that I currently write to a file on an SD card. I don't need it to be JSON, but everything else is, so it would be nice if it was too. 

Is there a way to start the serialization to the stream, and then add the data to the stream as it comes in. Otherwise, I need to allocate about 32K of memory to fit the whole file in the JSON library before writing it to the file.

bblanchon, 2019-03-04:
Hi @capricorn-one,

Thank you for this suggestion.
However, storing 2000 floats doesn't sound like a typical use case for JSON.
If I were you, I would simply dump the array as binary.

To keep the library easy to use for a large audience, I must sometimes say *no* to some requests.
Unfortunately, this feature request falls into that category: the use-case it too edgy, it would require too much work for too few users.

Thank you for your understanding, and thank you for using ArduinoJson.

Regards,
Benoit

closed 2019-03-04 by bblanchon

capricorn-one, 2019-03-04:
Hi Benoit,

Completely understand. Thank you for taking the time to consider and respond. I realized, one nice thing about JSON formatting is you can just append another object to the file, so I can manually write the correct syntax for the 2k floats, then use your fantastic library for the meta data.

Thanks again!

-Colin