bblanchon/ArduinoJson #917: Serialize JsonObject inside NestedArray

Findus76, 2019-03-01:
Hi
Thanks for a really good function.

I want to create a Jsonobject inside a nestedArray, but can't figure out how to do.
Exampel
`{
   "Type": "SystemFunction",
   "Function list": [
   {
      "Name": "FrontLight",
      "Type": "Light"
   },
   {
      "Name": "GabelLight",
      "Type": "Light"
   },
   {
      "Name": "GardenLight",
      "Type": "Light"
   }
]
}`

Something like this I think
`
StaticJsonDocument<SizeJSONmessageBuffe> doc;
JsonObject root = doc.to<JsonObject>();

root["Type"] = "SystemFunction";

JsonObject FrontLight = ??.createNestedObject();
FrontLight["Name"] = "FrontLight";
FrontLight["Type"] = "Light";

JsonObject GabelLight = ??.createNestedObject();
GabelLight["Name"] = GabelLight";
GabelLight["Type"] = "Light";

JsonObject GardenLight = ??.createNestedObject();
GardenLight["Name"] = "GardenLight";
GardenLight["Position"] = "OutdoorExtension Ventilation";

JsonArray function = root.createNestedArray("Function list");
function.add(FrontLight);
function.add(GabelLight);
function.add(GardenLight);`

I'am completely wring?
Brg
Findus

bblanchon, 2019-03-04:
Hi Findus,

Did you try the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/?utm_source=github&utm_medium=issues)?
Here is the code it generates for your JSON document:

```c++
const size_t capacity = JSON_ARRAY_SIZE(3) + 4*JSON_OBJECT_SIZE(2);
DynamicJsonDocument doc(capacity);

doc["Type"] = "SystemFunction";

JsonArray Function_list = doc.createNestedArray("Function list");

JsonObject Function_list_0 = Function_list.createNestedObject();
Function_list_0["Name"] = "FrontLight";
Function_list_0["Type"] = "Light";

JsonObject Function_list_1 = Function_list.createNestedObject();
Function_list_1["Name"] = "GabelLight";
Function_list_1["Type"] = "Light";

JsonObject Function_list_2 = Function_list.createNestedObject();
Function_list_2["Name"] = "GardenLight";
Function_list_2["Type"] = "Light";

serializeJson(doc, Serial);
```

Regards,
Benoit

Findus76, 2019-03-05:
Fantastic, what a great tool!

Thanks!
Brg
Findus

closed 2019-03-05 by Findus76