bblanchon/ArduinoJson #920: Question | how to convert a JSON to uint16_t

GiampaMan, 2019-03-04:
Hello!

I'm a beginner, I will appreciate all your help.

I'm trying to do an IR controller device with ESP8266. Due really long codes, ESP memory is failing, so I want to save a file with IR raw codes (really long raw codes) on SPIFFS using JSON library to format document.
My question is how do you think can be the best way to put an IR raw code in JSON format, and how can I read that array back in uint16_t, because IR library needs IR raw to be on that type of variable.

I will send an example of raw IR code as library needs:

`uint16_t powerOff[439] = {3480, 1760, 400, 444, 424, 1308, 420, 444, 424, 444, 424, 440, 424, 440, 424, 444, 420, 444, 420, 444, 424, 444, 448, 444, 420, 444, 420, 448, 420, 1332, 400, 444, 420, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 420, 1332, 400, 1332, 396, 1312, 420, 444, 452, 440, 424, 1308, 420, 448, 420, 444, 420, 444, 420, 448, 420, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 448, 444, 420, 444, 424, 444, 420, 444, 420, 444, 420, 448, 420, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 420, 444, 420, 448, 448, 440, 424, 444, 420, 444, 420, 1336, 396, 1332, 400, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 10008, 3484, 1736, 404, 440, 424, 1308, 448, 448, 420, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 444, 420, 444, 420, 1332, 400, 444, 424, 440, 452, 444, 420, 444, 420, 444, 424, 440, 424, 444, 424, 1304, 424, 1332, 400, 1308, 424, 440, 424, 444, 420, 1332, 400, 444, 424, 440, 424, 440, 424, 444, 424, 440, 448, 444, 424, 440, 424, 444, 424, 440, 424, 444, 420, 444, 424, 440, 424, 444, 424, 440, 424, 440, 424, 444, 424, 1308, 420, 1308, 424, 1308, 424, 440, 452, 440, 424, 440, 428, 440, 424, 1308, 424, 1304, 424, 444, 424, 1304, 424, 444, 424, 440, 428, 440, 424, 440, 428, 436, 428, 436, 428, 440, 428, 436, 456, 436, 428, 1304, 424, 1308, 424, 440, 428, 436, 428, 440, 428, 1300, 428, 1304, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 452, 440, 424, 440, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 424, 440, 428, 440, 424, 1308, 424, 1300, 432, 1304, 452, 440, 424, 440, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 1328, 404, 1304, 428, 1304, 424, 440, 428, 436, 428, 440, 452, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 428, 436, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 1304, 452, 440, 428, 436, 428, 1304, 428, 436, 428, 440, 424, 440, 428, 1304, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 452, 440, 428, 436, 428, 436, 428, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 428, 1300, 428, 440, 424, 440, 428, 1300, 428, 440, 428, 436, 456, 1324, 408};`

bblanchon, 2019-03-05:
Hi @GiampaMan,

I would not use a JSON document; instead, I would use a binary file.
Simply call [`File::readBytes()`](https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadbytes/) to read several bytes in one call.

Regards,
Benoit.

GiampaMan, 2019-03-05:
Hi!

Thanks for your response and for be interested in my question.
Do you mean a file with the numbers separated with ","? I have the readBytes function because when I read SPIFFS I put that data on a buffer. Is a great idea to simply read in one call, thanks for that. Do you know how to pass then that data to uint16_t?

I appreciate your help. Thanks!
Giampaolo Maneri.

jeot, 2019-03-05:
In your case, I would suggest a simple approach. After reading all your data from the file into a buffer of bytes, create another buffer of uint16_t and fill it using the previous buffer:
```
byte buffer[439*2]; // fill the buffer from file

uint16_t powerOff[439];
for (int i = 0; i != 439; i++)
{
  powerOff[i] = buffer[2*i]+256*buffer[2*i+1];
}
```

GiampaMan, 2019-03-06:
Hi @jeot ! 
Thanks for your response! I tested, but have some issues. What do you think about my code? Is right?

```
#include <Arduino.h>
#include <FS.h>
#include <stdint.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#define IR_LED 4

IRsend irsend(IR_LED); 

void initModule()
{
    SPIFFS.begin();

    
    Serial.println("Mounting FS...");
    if (!SPIFFS.begin()) 
    {
        Serial.println("Failed to mount file system");
        return;
    }
}

bool loadPowerOff() 
{
    File powerOff = SPIFFS.open("/powerOff", "r");
    if (!powerOff) 
    {
        Serial.println("Failed to open powerOff file");
        return false;
    }

    size_t size = powerOff.size();

    std::unique_ptr<char[]> buf(new char[size]);
    powerOff.readBytes(buf.get(), size);
    buf[439*2]; // fill the buffer from file
    uint16_t powerOff[439];
    for (int i = 0; i != 439; i++)
    {
        powerOff[i] = buf[2*i]+256*buf[2*i+1];
    }

    irsend1.sendRaw(powerOff,439,38);

    return true;
}

void setup() {
  initModule();
}

void loop() {
  loadPowerOff();
  }
```

jeot, 2019-03-06:
Hi @GiampaMan,
First of all, you don't have `Serial.begin(9600)`. I don't know if `Serial.println(...)` work without it or not!
You have `powerOff ` both for File and the buffer. Change the variable names.
I don't undrestant `std::unique_ptr<char[]> buf(new char[size]);`. you can simpy use `uint8_t buf[size];`
change the variable names and debug your code by checking the values of variables with `Serial.println(...)` in your console log.
It should look something like this:
```
void initModule()
{
    SPIFFS.begin();
    Serial.begin(9600);
    
    Serial.println("Mounting FS...");
    if (!SPIFFS.begin()) 
    {
        Serial.println("Failed to mount file system");
        return;
    }
}

bool loadPowerOff() 
{
    File file_id= SPIFFS.open("/powerOff", "r");
    if (!file_id) 
    {
        Serial.println("Failed to open powerOff file");
        return false;
    }

    size_t size = file_id.size();

    uint8_t buf[size];
    file_id.readBytes(buf, size);
    
    uint16_t powerOff[439];
    for (int i = 0; i != 439; i++)
    {
        powerOff[i] = buf[2*i]+256*buf[2*i+1];
    }
    // show the data in console and check if it is correct:
    for (int i = 0; i != 439; i++)
    {
        Serial.print(powerOff[i]); Serial.print(", ");
    }
    irsend1.sendRaw(powerOff,439,38);

    return true;
}
```

GiampaMan, 2019-03-07:
Hi @jeot !
Thanks for your help. 
I was checking on serial port, problem is that are not the correct values.
I think this is related to the "for" loop:
```
    for (int i = 0; i != 439; i++)
    {
        powerOff[i] = buf[2*i]+256*buf[2*i+1];
    }
```
This is the part: `powerOff[i] = buf[2*i]+256*buf[2*i+1];`
Depending on number that multiply "buf" I get different values. When I put `powerOff[i] = buf[2*i]+256*buf[2*i+1];` ESP crash and I get higher values, but if I instead put `powerOff[i] = buf[2*i]+buf[2*i+1];` numbers printed are slower and ESP don't crash. Both are not the original numbers on file. Which is the correct number to exactly read the file originally? Or which is the configuration I need to use?
The first number I get on serial using `powerOff[i] = buf[2*i]+256*buf[2*i+1];` is: 13363 but correct one is: 3480.
Another question is (only to know, I did not tested this yet), what happens if I put a smaller IR code on SPIFFS file using same programming code you sent me, for example:
`uint16_t powerOff[59] = {3096, 9888, 492, 1608, 464, 580, 460, 576, 456, 576, 460, 1640, 460, 560, 484, 552, 468, 564, 484, 1604, 468, 1612, 464, 556, 488, 556, 484, 564, 484, 556, 488, 556, 488, 548, 484, 548, 488, 548, 488, 548, 484, 548, 488, 548, 484, 1604, 468, 552, 488, 1592, 496, 544, 488, 548, 484, 548, 488, 1608, 468};`

The limit is 439? Or the logic on program is affected? In that case is possible to use a dynamic math operation process without specifying number?

I changed something to your example because compiler don't wanted to compile. This is the code I used to tell you my experience:

```
bool loadPowerOff() 
{
    File file_id= SPIFFS.open("/powerOff", "r");
    if (!file_id) 
    {
        sismaSerial->println("Failed to open powerOff file");
        return false;
    }

    size_t size = file_id.size();
    //uint8_t buf[size]; // uint8_t is same as byte
    std::unique_ptr<char[]> buf(new char[size]);
    file_id.readBytes(buf.get(), size);
    uint16_t powerOff[439];
    for (int i = 0; i != 439; i++)
    {
        powerOff[i] = buf[2*i]+256*buf[2*i+1];
    }
    // show the data in console and check if it is correct:
    for (int i = 0; i != 439; i++)
    {
        sismaSerial->print(powerOff[i]); 
        sismaSerial->print(", ");
    }
    irsend1.sendRaw(powerOff,439,38);
    return true;
}
```
This line is the one having problems: `//uint8_t buf[size]; // uint8_t is same as byte` 
Instead I used:     `std::unique_ptr<char[]> buf(new char[size]);`

Thanks you so much about your help and I hope you can help me with my questions.

Regards,
Giampaolo Maneri.

jeot, 2019-03-07:
@GiampaMan 
Try this: `powerOff[i] = 256*buf[2*i]+buf[2*i+1]`
439 is not the limit. I used this number as an example because in your first question you used 439!

jeot, 2019-03-07:
@GiampaMan 
Also, I recommend posting your question on [StackOverflow](https://stackoverflow.com/)
This is not a good place for these types of question and many more people can help you there!

GiampaMan, 2019-03-08:
Hi @jeot!

I tried `powerOff[i] = 256*buf[2*i]+buf[2*i+1];` now I get on serial port as first value: 13108. Still incorrect value. What do you think could be another line to test?
Thanks for the recommendation about using Stack Overflow, I will investigate about that page.

Thanks a lot for being interested with my problem. I will wait for your feedback. 

Regards,
Giampaolo Maneri

jeot, 2019-03-08:
@GiampaMan
I am only one person and I don't know about SPIFFS and how it works, therefore I recommended StackOverflow to you.
It should be a problem of saving and reading from file. How did you save the data into the file?

bblanchon, 2019-03-08:
Hi @GiampaMan,

It looks like an [endianness](https://en.wikipedia.org/wiki/Endianness) issue.
I'm pretty sure that ESP8266 uses big endian, so make sure you save the file with the same convention.

Also, you don't need to duplicate the buffer to go from `uint8_t` to `uint16_t`; a simple cast is enough.
Then, assuming that there is no zero in the file, we can use `readString()` to simplify the memory management:

```c++
bool loadPowerOff() 
{
    File file= SPIFFS.open("/powerOff", "r");
    String fileContent = file.readString();
    file.close();
     
    auto powerOff = (const uint16_t*)fileContent._str();
   for (int i = 0; i != 439; i++)
    {
        sismaSerial->print(powerOff[i]); 
        sismaSerial->print(", ");
    }

   //...
}
```

Alertnatively, you could avoid the `String` if you do the read/send repeatedly:

```c++
bool loadPowerOff() 
{
    File file= SPIFFS.open("/powerOff", "r");

   for (int i = 0; i != 439; i++)
    {
        uint16_t x;
        file.readBytes(&x, sizeof(x));
        sismaSerial->print(powerOff[i]); 
        sismaSerial->print(", ");
    }

    file.close();
    //...
}
```

I'll now close this issue because it's not related to ArduinoJson.
I hope that the information we gave will help you with your project.
If not, please follow @jeot's advice and post your question on StackOverflow.

Thank you for your understanding.

Regards,
Benoit

closed 2019-03-08 by bblanchon

GiampaMan, 2019-03-08:
Hi @bblanchon @jeot 
Thanks for all your help.

I created a StackOverflow Account and asked there. Here is the link if you want to go there and continue helping:
https://stackoverflow.com/q/55069758/11172629?sem=2

I tested your methods @bblanchon , the one using strings gave me wrong numbers, the other not compiled.

@jeot this is the content of the file on SPIFFS: `3480, 1760, 400, 444, 424, 1308, 420, 444, 424, 444, 424, 440, 424, 440, 424, 444, 420, 444, 420, 444, 424, 444, 448, 444, 420, 444, 420, 448, 420, 1332, 400, 444, 420, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 420, 1332, 400, 1332, 396, 1312, 420, 444, 452, 440, 424, 1308, 420, 448, 420, 444, 420, 444, 420, 448, 420, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 448, 444, 420, 444, 424, 444, 420, 444, 420, 444, 420, 448, 420, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 444, 420, 444, 420, 448, 448, 440, 424, 444, 420, 444, 420, 1336, 396, 1332, 400, 444, 420, 444, 424, 444, 420, 444, 420, 444, 424, 10008, 3484, 1736, 404, 440, 424, 1308, 448, 448, 420, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 440, 424, 444, 420, 444, 424, 444, 420, 444, 420, 1332, 400, 444, 424, 440, 452, 444, 420, 444, 420, 444, 424, 440, 424, 444, 424, 1304, 424, 1332, 400, 1308, 424, 440, 424, 444, 420, 1332, 400, 444, 424, 440, 424, 440, 424, 444, 424, 440, 448, 444, 424, 440, 424, 444, 424, 440, 424, 444, 420, 444, 424, 440, 424, 444, 424, 440, 424, 440, 424, 444, 424, 1308, 420, 1308, 424, 1308, 424, 440, 452, 440, 424, 440, 428, 440, 424, 1308, 424, 1304, 424, 444, 424, 1304, 424, 444, 424, 440, 428, 440, 424, 440, 428, 436, 428, 436, 428, 440, 428, 436, 456, 436, 428, 1304, 424, 1308, 424, 440, 428, 436, 428, 440, 428, 1300, 428, 1304, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 452, 440, 424, 440, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 424, 440, 428, 440, 424, 1308, 424, 1300, 432, 1304, 452, 440, 424, 440, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 1328, 404, 1304, 428, 1304, 424, 440, 428, 436, 428, 440, 452, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 428, 436, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 1304, 452, 440, 428, 436, 428, 1304, 428, 436, 428, 440, 424, 440, 428, 1304, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 452, 440, 428, 436, 428, 436, 428, 440, 428, 436, 428, 440, 424, 440, 428, 436, 428, 440, 428, 1300, 428, 440, 424, 440, 428, 1300, 428, 440, 428, 436, 456, 1324, 408`
Is exactly like that.

Thanks for all!