bblanchon/ArduinoJson #922: Add binary support for MessagePack (C

jeot, 2019-03-05:


jeot, 2019-03-05:
I accidentally hit the Enter key while typing! sorry about that!

the full implementation of MessagePack is available here:
https://github.com/HEADS-project/arduino_msgpack

The only issue is that it only works with streams. I really like the versatility of ArduinoJson (working with Streams, Strings, buffers, stdio, ...) and would like to see the full implementation of MessagePack in future updates of ArduinoJson. 
Thanks for your wonderful work.

jeot, 2019-03-05:
Example of the mentioned issue:
The MessagePack code (total bytes: 79):

0x84 0xA7 0x56 0x65 0x72 0x73 0x69 0x6F 
0x6E 0xC4 0x03 0x01 0x01 0x01 0xA8 0x44 
0x61 0x74 0x65 0x54 0x69 0x6D 0x65 0x00 
0xA4 0x54 0x79 0x70 0x65 0x01 0xAE 0x41 
0x75 0x74 0x68 0x65 0x6E 0x74 0x69 0x63 
0x61 0x74 0x69 0x6F 0x6E 0x82 0xA6 0x53 
0x65 0x72 0x69 0x61 0x6C 0xC4 0x08 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x01 0xAB 
0x53 0x6F 0x66 0x74 0x56 0x65 0x72 0x73 
0x69 0x6F 0x6E 0xC4 0x02 0x00 0x00 

Serialize the message pack to json with HEADS's arduino_msgpack library:
{"Version":  0x1 0x1 0x1', "DateTime": 0, "Type": 1, "Authentication": {"Serial":  0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1', "SoftVersion":  0x0 0x0'}}

Trying to serialize with ArudinoJson v6.9.1:
deserializeMsgPack() failed: NotSupported

LioRei, 2022-08-30:
Hi,
if you want some help on this topic I could share the code of my library as I need it for it ;)

From now, I only have done the serialization part.

commit ae0136da (2024-04-29): Add MsgPack bin8/bin16/bin32 support

Closes #2078
Closes #922
commit 18a9a5b5 (2024-04-29): Add MsgPack bin8/bin16/bin32 support

Closes #2078
Closes #922
closed 2024-04-29 by bblanchon