bblanchon/ArduinoJson #923: Function that returns any JsonVariant no longer works in ArduinoJson6

BryceBeagle, 2019-03-06:
In ArduinoJson 5, we used to define a function that returned any JSON-encodable type kinda like this:

```c++
std::function<JsonVariant()> coolFunction = [&] {return 5;};
```

At some point (probably in ArduinoJson 6), JsonVariant was changed and this no longer works:
```
error: conversion from 'setup()::__lambda0' to non-scalar type 'std::function<ArduinoJson691_000::VariantRef()>' requested
```

Is there a way to express this now, perhaps using another type?

bblanchon, 2019-03-07:
Hi @BryceBeagle,

In ArduinoJson 6, [JsonVariant](https://arduinojson.org/v6/api/jsonvariant/) is a reference like [JsonArray](https://arduinojson.org/v6/api/jsonarray/) and [JsonObject](https://arduinojson.org/v6/api/jsonobject/).
If you want a variant with value semantics, you must use a [JsonDocument](https://arduinojson.org/v6/api/jsondocument/)

However, I find your example very farfetched.
If I were you, I would refactor to something simpler.

Regards,
Benoit

BryceBeagle, 2019-03-11:
Sorry, I'm not sure I understand what you mean about using JsonDocument? Simply replacing `JsonVariant` with with `JsonDocument` does not work?

Now, obviously the lambda function there is pretty worthless, but it was meant to be just an example. I'm simply looking to store references to functions that can return _any_ value that is of a Json serializable type. JsonVariant sufficed here before. 

Thanks for the help.

formerredbeard, 2019-03-12:
BryceBeagle, there were major structural changes between 5 and 6. Check out this from his site: https://arduinojson.org/v6/doc/upgrade/ 
I just refactored my code to use it. You can see an example of the new code at: https://arduinojson.org/v6/assistant/

bblanchon, 2019-03-13:
Hi @BryceBeagle,

If you want to follow this route, you need to do that:

```c++
std::function<DynamicJsonDocument()> uncoolFunction = []() {
  DynamicJsonDocument doc(16);
  doc.set(5);
  return doc;
};
```

Again, if I were you I would refactor the code to avoid this kind of construct.

Regards,
Benoit

closed 2019-03-23 by bblanchon