bblanchon/ArduinoJson #924: Help DynamicJsonDocument long json

clydde, 2019-03-06:
Hello, I write the message with googleTranslator, so I hope I am sorry if I do not understand too much what I want to express.

I would like to begin by saying that I have read the documentation and looked at the examples of the book, but I still do not understand how the library works. The book I bought refers to v 5 and I am looking at the document of the 6, this increases the confusion.

We're there ....

Use Arduino one, with a microSD expansion module and your ArduinoJson library v6. Now I am doing tests with a small json, but the final version should contain a very large json.

The problem I have is that I do not understand the good work done by ArduinoJson to read the json of SD.

The document DynamicJsonDocument method (); It works fine if it's a small file and you know what measure it will take, but in my case I will have a very big json and I will not know what measure it will take.

I'm going to put here my code that works, but it does not work for me.

`#include <ArduinoJson.h>
#include <SD.h>
#include <SPI.h>

void setup() {
  Serial.begin(9600);
  while (!Serial) continue;
  Serial.println("Read SD...");
  if (!SD.begin(4)) {
    Serial.println("Error  SD!");
    while (1);
  }
  Serial.println("SD ready.\n");

  File  myFile = SD.open("config.txt", FILE_READ);

  DynamicJsonDocument doc(455); // Return partial json content
 DynamicJsonDocument doc(810); // ReturnFailed  No memory

  deserializeJson(doc, myFile);

  for (JsonObject repo : doc.as<JsonArray>()) {
    Serial.println(repo["tar"].as<char*>());
    Serial.println(repo["pin"].as<char*>());
  }

  DeserializationError error = deserializeJson(doc, myFile);

  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
}

void loop() {

}`

Example My test small JSON File: // The final are 4000 keys
`[
    {
        "tar": "f11hf0876fg0h2871n657465er4476h1",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f22hf0876fg0h2871n657465er4476h2",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f33hf0876fg0h2871n657465er4476h3",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f11hf0876fg0h2871n657465er4476h1",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f22hf0876fg0h2871n657465er4476h2",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f11hf0876fg0h2871n657465er4476h1",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f22hf0876fg0h2871n657465er4476h2",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f11hf0876fg0h2871n657465er4476h1",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    },
    {
        "tar": "f22hf0876fg0h2871n657465er4476h2",
        "pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
    }
]`

Can someone help me? What am I leaving?

Thank you very much!

xEvi

bblanchon, 2019-03-07:
Hi @clydde,

First of all, thank you for purchasing the book.
If you purchased the ebook, you can download the new edition for free; you just need to go to the download page. If you purchased the printed edition, please email me and we'll find a solution.

Your problem is:

1. The Arduino UNO has very few RAM (only 2KB)
2. The input document is very large (needs more than 300KB)

If you want to keep the Arduino UNO, your only option is to deserialize the JSON document in chunks.
In the book, I demonstrate this technique for OpenWeatherMap, Weather Underground, (old edition) and Reddit (new edition).

I created a FAQ entry for that: [How to deserialize a very large document?](https://arduinojson.org/v6/faq/how-to-deserialize-a-very-large-document/)
Please have a look.

Regards,
Benoit

clydde, 2019-03-07:
Thank you,  I just downloaded the book again, this time the v6, are you very kind to offer the update for free.

As far as my problem is concerned, although I still can not solve it, I am glad to know that it is possible to do it, I will study hard to achieve it!

Thank you very much!

clydde, 2019-03-08:
.... imagine that I get an Arduino Mega (Flash 256K), do you know if it would be possible to copy a json file of for example 50kb in Flash memory and work from there?

The idea is The code checks if there is a json in the SD, if it is "true" it is copied in the Flash ....

This may require participating in the Flash, which I do not know if it is possible?

All this madness occurs to me thinking (less hadware = less problems),

What do you think?

bblanchon, 2019-03-08:
Hi @clydde,

No, I don't think using the Flash is a good idea, and I'm not even sure it would work.

Your JSON document is very strange: it looks like a binary format.
If I were you, I would reconsider the file format because I don't think JSON is a good fit.
If you can't change the JSON format, use the "parse in chunks" technique, it's easier than it looks.

Regards,
Benoit

clydde, 2019-03-08:
Thanks for your comments!

The project I am doing consists of multiple RFID accesses with double authentication, in the key "tar" I keep the number of the card and in "pin" a 4-digit code. The keys are long because they are encrypted.

One of the demands of the project is that the system can function in an autonomous way.

In an example of the book, he puts DynamicJsonDocument (16 * 1024)
and in something similar to StaticJsonDocument (300)

What do you recomend? Where did you know the values ​​16 * 1024?

Thank you!

bblanchon, 2019-03-20:
You're welcome @clydde.

closed 2019-03-20 by bblanchon