bblanchon/ArduinoJson #925: Cannot get numeric values from JsonObject

gsloan-chef, 2019-03-07:
This is the JSON string I'm trying to work with:
{"Kitchen Speaker": {"alarm":[],"timer":[{"fire_time":1551912503000.0,"id":"timer/5c8ac0d5-0000-2ec5-a940-f4f5e80dd9b0","original_duration":0.0,"status":1},{"fire_time":1551912331000.0,"id":"timer/5c8dfdfe-0000-279d-8b42-883d24fd518c","original_duration":0.0,"status":1}]}, "Living Roomm speaker": {"alarm":[],"timer":[{"fire_time":1551912996000.0,"id":"timer/5c8dd97c-0000-2023-b66f-94eb2c145af0","original_duration":0.0,"status":1}]}}

I'm trying to pull, for example, ["Kitchen Speaker"]["timer"][0]["fire_time"]

I used the Assistant and copy-pasted code from there. No matter what I do, the value comes back as "0" (or 0.0 if I'm trying to pull it as a double).  Same for "status".  String data works fine (e.g., I'm able to get "id" as a const char* no problem).  Here's my code

`
#include <ArduinoJson.h>

String testData = "{\"Kitchen Speaker\": {\"alarm\":[],\"timer\":[{\"fire_time\":551912503000.0,\"id\":\"timer/5c8ac0d5-0000-2ec5-a940-f4f5e80dd9b0\",\"original_duration\":0.0,\"status\":1},{\"fire_time\":1551912331000.0,\"id\":\"timer/5c8dfdfe-0000-279d-8b42-883d24fd518c\",\"original_duration\":0.0,\"status\":1}]}, \"Living Roomm speaker\": {\"alarm\":[],\"timer\":[{\"fire_time\":1551912996000.0,\"id\":\"timer/5c8dd97c-0000-2023-b66f-94eb2c145af0\",\"original_duration\":0.0,\"status\":1}]}}";

const size_t CAPACITY = 2*JSON_ARRAY_SIZE(0) + 2*JSON_ARRAY_SIZE(2) + 3*JSON_OBJECT_SIZE(2) + 4*JSON_OBJECT_SIZE(4);

void setup() {
Serial.begin(115200);
}

void loop() {
StaticJsonDocument<CAPACITY> doc;
  deserializeJson(doc, testData);
  JsonObject Kitchen_Speaker_timer_0 = doc["Kitchen Speaker"]["timer"][0];
  long Kitchen_Speaker_timer_0_fire_time = Kitchen_Speaker_timer_0["fire_time"];
  Serial.print("fire time: ");
  Serial.println(Kitchen_Speaker_timer_0_fire_time);
  delay(2000);
}`

gsloan-chef, 2019-03-07:
Well I seem to have a partial answer - for the "fire_time" it's an out of range issue. The value is too large for a long.

But float/double should be able to handle it. 

That also doesn't explain why I'm getting 0 with this:
```
JsonObject Kitchen_Speaker_timer_0 = doc["Kitchen Speaker"]["timer"][0];
int Kitchen_Speaker_timer_0_fire_time = Kitchen_Speaker_timer_0["status"];
```

bblanchon, 2019-03-08:
Hi @gsloan-chef,

The [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) is too small.
If you check the result of [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/), you'll see that it returns [NoMemory](https://arduinojson.org/v6/api/misc/deserializationerror/).

Indeed, when the input is read-only, ArduinoJson needs additional space in the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) to copy the strings.

The [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) says you need at least 304 bytes to store the strings.
Look at the program generated by the Assistant, and you'll see that it adds 340 bytes to the capacity of the JsonDocument:

```c++
const size_t capacity = /*...*/ + 340;
DynamicJsonDocument doc(capacity);
```

Regards,
Benoit

closed 2019-03-23 by bblanchon