bblanchon/ArduinoJson #926: Compile Error on Particle IDE version 5.13.5

formerredbeard, 2019-03-08:
The code below compiles on Arduino IDE but not Particle IDE using version 5.13.5. (Granted once Vers6 is available on Particle.io, I probably won't care unless it has a similar error) If I change the output to be Output to String format, I get a different error, but still an error on Particle IDE.
----------------------------------------------------
On Particle IDE get these among other related errors:
lib/ArduinoJson/ArduinoJson/Deserialization/JsonParser.hpp:73:50: no type named 'Reader' in 'struct ArduinoJson::Internals::StringTraits<String, void>'

testarduinojson.ino:56:31: no matching function for call to 'ArduinoJson::JsonVariant::as()'

lib/ArduinoJson/ArduinoJson/Deserialization/../JsonVariant.hpp:150:3: no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, String>'

Code (complete)-----------------------------------------
`
// This #include statement was automatically added by the Particle IDE.
#include <ArduinoJson.h>

String test;
int data1;
float data2;

void setup() {
 //Particle.variable("test", test);
// Particle.variable("data1", data1);
 //Particle.variable("data2", data2);
 
}

void loop() {
test1();
test = test2();

}

void test1(){
   const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(3) + 60;
DynamicJsonBuffer jsonBuffer(capacity);

String json = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.75608,2.302038]}";
//strcpy(test,json);
test=json;

JsonObject& root = jsonBuffer.parseObject(test);

const char* sensor = root["sensor"]; // "gps"
long time = root["time"]; // 1351824120

float data_0 = root["data"][0]; // 48.75608
float data_1 = root["data"][1]; // 2.302038
data1=data_0;
data2=data_1;

}

String test2(){
const size_t capacity = JSON_ARRAY_SIZE(2) + JSON_OBJECT_SIZE(3);
DynamicJsonBuffer jsonBuffer(capacity);

JsonObject& root = jsonBuffer.createObject();
root["sensor"] = "gps";
root["time"] = 1351824120;

JsonArray& data = root.createNestedArray("data");
data.add(48.75608);
data.add(2.302038);

 JsonVariant v = root;
// Cast the JsonVariant to a string
 String output = v.as<String>();
 return output;
}

`

bblanchon, 2019-03-08:
Hi @formerredbeard,

Thank you for reporting this issue.

This error message means that ArduinoJson doesn't recognize the `String` class.
ArduinoJson enables the support of the `String` class when the `ARDUINO` symbol is defined.
I don't know the details, but I remember that Particle doesn't define `ARDUINO` unless you use some kind of compatibility mode.

To fix this issue, you must force the support of the `String` class by defining [`ARDUINOJSON_ENABLE_ARDUINO_STRING`](https://arduinojson.org/v6/api/config/enable_arduino_string/) to 1:

```c++
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1
#include <ArduinoJson.h>
```

Please confirm if it works, so I can put that in the FAQ.
Alternatively, I agree to modify the library if there is a better way to detect the availability of the `String` class.

BTW, I'll push version 6 to Particle in a moment, but the problem will be the same.

Regards,
Benoit

formerredbeard, 2019-03-08:
Nailed it! Thanks. I had read about that months ago and forgot it. All that is needed is the following:

#include "Arduino.h"

Which I added and just tested. It works great!

closed 2019-03-08 by formerredbeard

kentkeller, 2019-03-08:
Well I added both of the lines as you suggested, but the one include Arduino.h fixes the compile error.:-)