bblanchon/ArduinoJson #927: no type named 'const_iterator' in 'class ArduinoJson691_000::ObjectRef'

JeffrySteegmans, 2019-03-10:
I just update from version 5.x to 6.x and now I get the following error in the file ArduinoJson/Deserialization/IteratorReader.cpp on line 27.

[image]

Any solutions?

bblanchon, 2019-03-10:
Hi @JeffrySteegmans,

I'm going to need a bit more context...

JeffrySteegmans, 2019-03-10:
I'm building with VS code and platformIO. Also installed the update with platformIO.
I changed my code according to the upgrade wizzard but this is an error within the library i think.

This is my complete build log: https://pastebin.com/VKWreL7H

bblanchon, 2019-03-10:
It seems that the call to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/) is incorrect.
Can you verify?

JeffrySteegmans, 2019-03-14:
Damn you're right. I did something wrong when converting from 5.x to 6.x.
Tx for the help!!

closed 2019-03-14 by JeffrySteegmans