bblanchon/ArduinoJson #928: How can i convert a json document to a const char* that is needed in the MQTT library PubSubClient( v2,7) ?

NicklasP, 2019-03-12:
How can i convert a json document to a const char* that is needed in the MQTT library PubSubClient( v2,7) ?  Im using a ESP8266 on a nodeMCU board

bblanchon, 2019-03-12:
Hi @NicklasP,

You must allocate a buffer and serialize the JSON document to this buffer:

```c++
char buffer[256];
serializeJson(doc, buffer);
client.publish("topic", buffer);
```

I never used this library, so my example might be clumsy.
There are plenty of examples on the Internet if you need.

Regards,
Benoit

NicklasP, 2019-03-13:
Thanks for your answer and all your hard work
I solved it eventually like this.
 const int capacity = JSON_OBJECT_SIZE(3);
  StaticJsonDocument<capacity> doc;
  doc["TEMP1"] = String(teststring1);
  doc["DTHTemp"] = DHT22_temp;
  doc["HUM"] = String("33.4");
  char payload[capacity + 2];  //+2 ? not sure about that 
  serializeJson(doc,payload);
client.publish("MyTopic",payload);

Im using it to do some thing like this:
https://youtu.be/JdV4x925au0

closed 2019-03-13 by NicklasP

bblanchon, 2019-03-13:
Cool!

Do not confuse the capacity of the [`JsonDocument`](http://127.0.0.1:4000/v6/api/jsondocument/), which is the RAM required to store the data structure, with the capacity of `payload`, which is the length of the JSON string.

If you accept `String` in your code, you can write:

```c++
String payload;
serializeJson(doc,payload);
client.publish("MyTopic", payload.c_str());
```

But remember that `String` produces [heap fragmentation](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html).

bblanchon, 2019-03-17:
I added the following entry in the FAQ:

* [How to use ArduinoJson with PubSubClient, the MQTT library?](https://arduinojson.org/v6/faq/how-to-use-arduinojson-with-pubsubclient/)

Let me know if you have any comment.

NicklasP, 2019-03-17:
Hi

Brilliant idea to add it to the FAQ. Thank you Mr. Blanchon.

Regards
Nicklas