bblanchon/ArduinoJson #930: Needed to increase the JSON_OBJECT_SIZE by 1 to get correct outcome 6.9.1

formerredbeard, 2019-03-12:


formerredbeard, 2019-03-12:
On ParticleIO at least, I had to increase the JSON_OBJECT_SIZE by 1 to get the correct result below, JSON_OBJECT_SIZE(8) needed to be used instead of the Arduino Assistant recommended (7) 

Example:
void serializeSite(int currentTemp) {
  const size_t capacity =
      JSON_OBJECT_SIZE(8); 
  DynamicJsonDocument docSite(capacity);

  docSite["name"] = site.name;
  docSite["HTemp"] = site.highTemp;
  docSite["LTemp"] = site.lowTemp;
  docSite["unitTemp"] = site.unitTemp;
  docSite["CTemp"] = currentTemp;
  docSite["Probe"] = site.tempprobe;
  docSite["Number"] = NUMBER;
  serializeJson(docSite, output);
}

JeppeMariagerLam, 2019-03-13:
I'm seeing a similar issue, although I have to increase more than just by 1...

bblanchon, 2019-03-13:
Hi @formerredbear,
Hi @JeppeMariagerLam,

As you can see [on this demo](https://wandbox.org/permlink/4JOMGU8YWufnFirF), `JSON_OBJECT_SIZE(7)` is sufficient for your example.

However, ArduinoJson duplicates every `String` you save in the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) .
For example, if `site.name` is a `String`, the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) makes a copy of it; so it needs to be larger.

The [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) computes the size required to duplicate strings under the title "Additional bytes for strings duplication."
Simply add this value to the capacity of the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) and everything will be fine.

Regards,
Benoit

formerredbeard, 2019-03-13:
The following code when run on a Particle Photon has different outputs for the two variables: output1 and output2. Basically when I used variables instead of the values directly generated by the Arduino Assistant, things changed. I did not use a String thanks to your book. :-)

`//Test to confirm Arduino JSON not working as expected

// This #include statement was automatically added by the Particle IDE.
#include "Arduino.h"
#include <ArduinoJson.h>
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1
#define NUMBER 3
// Site Variables
struct SiteStored {  
  char name[16];
  int highTemp;    
  int lowTemp;    
  int unitTemp;    // C 1 or F 2
  bool tempprobe;  // Manually Disable temperature probe
};

SiteStored site{"MySite123456789", 40, 1, 1, true};
char output1[100];
char output2[100];

void setup() {
    Particle.variable("Site1", output1);
    Particle.variable("Site2", output2);
}

void loop(){
    double currentTemp=30;
    serializeSite1((int) currentTemp);
    serializeSite2();
     Particle.publish("Site1", output1, PRIVATE);
     Particle.publish("Site2", output2, PRIVATE);
     delay(10000);
}

void serializeSite1(int currentTemp) {
  const size_t capacity = JSON_OBJECT_SIZE(7); 
  DynamicJsonDocument doc(capacity);

  doc["name"] = site.name;
  doc["HTemp"] = site.highTemp;
  doc["LTemp"] = site.lowTemp;
  doc["unitTemp"] = site.unitTemp;
  doc["CTemp"] = currentTemp;
  doc["Probe"] = site.tempprobe;
  doc["Number"] = NUMBER;
  serializeJson(doc, output1);
}

void serializeSite2() {
  const size_t capacity = JSON_OBJECT_SIZE(7); 
  DynamicJsonDocument doc(capacity);

  doc["name"] = "MySite123456789";
  doc["HTemp"] = 40;
  doc["LTemp"] = 1;
  doc["unitTemp"] = 1;
  doc["CTemp"] = 30;
  doc["Probe"] = true;
  doc["Towers"] = 3;
  serializeJson(doc, output2);
}
`

formerredbeard, 2019-03-13:
The following code when run on a Particle Photon has different outputs for the two variables: output1 and output2. Basically when I used variables instead of the values directly generated by the Arduino Assistant, things changed. I did not use a String thanks to your book. :-)

`//Test to confirm Arduino JSON not working as expected

// This #include statement was automatically added by the Particle IDE.
#include "Arduino.h"
#include <ArduinoJson.h>
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1
#define NUMBER 3
// Site Variables
struct SiteStored {  
  char name[16];
  int highTemp;    
  int lowTemp;    
  int unitTemp;    // C 1 or F 2
  bool tempprobe;  // Manually Disable temperature probe
};

SiteStored site{"MySite123456789", 40, 1, 1, true};
char output1[100];
char output2[100];

void setup() {
    Particle.variable("Site1", output1);
    Particle.variable("Site2", output2);
}

void loop(){
    double currentTemp=30;
    serializeSite1((int) currentTemp);
    serializeSite2();
     Particle.publish("Site1", output1, PRIVATE);
     Particle.publish("Site2", output2, PRIVATE);
     delay(10000);
}

void serializeSite1(int currentTemp) {
  const size_t capacity = JSON_OBJECT_SIZE(7); 
  DynamicJsonDocument doc(capacity);

  doc["name"] = site.name;
  doc["HTemp"] = site.highTemp;
  doc["LTemp"] = site.lowTemp;
  doc["unitTemp"] = site.unitTemp;
  doc["CTemp"] = currentTemp;
  doc["Probe"] = site.tempprobe;
  doc["Number"] = NUMBER;
  serializeJson(doc, output1);
}

void serializeSite2() {
  const size_t capacity = JSON_OBJECT_SIZE(7); 
  DynamicJsonDocument doc(capacity);

  doc["name"] = "MySite123456789";
  doc["HTemp"] = 40;
  doc["LTemp"] = 1;
  doc["unitTemp"] = 1;
  doc["CTemp"] = 30;
  doc["Probe"] = true;
  doc["Number"] = 3;
  serializeJson(doc, output2);
}
`

formerredbeard, 2019-03-13:
And perhaps it is not a code error, but a tweak to the Arduino Assistant is necessary to add a few bytes buffer. I started with this JSON at Arduino Assistant to generate the code:
{"name":"MySite123456789","HTemp":40,"LTemp":1,"unitTemp":1,"CTemp":30,"Probe":true,"Number":3}

bblanchon, 2019-03-14:
Hi @formerredbeard,

I didn't mention it in my previous message but ArduinoJson duplicates `char*` as well. Only a `const char*` (typically string literals) is stored with a pointer.
That's why it makes a copy of `site.name`; so you need to increase the document size by 16.

The ArduinoJson Assistant doesn't include the "extra bytes" in the serialization program.
I'm not sure if I should include them or not...
It could add the bytes required to save the values, not the keys; but I fear it will be very confusing.

Regards,
Benoit

formerredbeard, 2019-03-14:
OK. Thanks. Perhaps add a '//+ sizeof of any char arrays"
to end of Arduino Assistant line for the capacity calculation.

Will amend my code. I plan to amend by putting in +sizeof(site.name) and other char arrays in other serialization functions. It may not be as efficient, however if I bump up the size of the arrays in other places in the future I won't need to come back and fix. 

Thanks so much!

bblanchon, 2019-03-15:
You're welcome.
Thank you for using ArduinoJson

closed 2019-03-15 by bblanchon