bblanchon/ArduinoJson #931: A typo in the new version of the book?

liuzheng1990, 2019-03-15:
Hi. I'm not sure whether this is the right place to report some typo in the book. I love the ArduinoJson module and the book. It seems there is a significant improvement by introducing JsonDocument. I like the idea to treat a JsonArray and a JsonObject in the same framework. 

However, I think there are some typo when the book was changed according to this change. 
On page 62 to 64, the JsonDocument (I think more or less equivalent to JsonBuffer in the previous versions) was called "jb":

```
StaticJsonDocument<capacity> jb;
```

and on Page 65, when we parse the Json data, a variable called `doc` suddenly appeared.

```
DeserializationError err = deserializeJson(doc, input);
```

In the beginning it puzzled me on what this `doc` was. After browsing back and forth for a couple of times, I didn't find the declaration of it. When I checked the "parsing Json Array" section, I realized maybe we should had

```
StaticJsonDocument<capacity> doc; //instead of jb?
```

Am I right? Pls correct me if I made it wrongly.

bblanchon, 2019-03-15:
Hi @liuzheng1990,

Thank you for purchasing the book and for reporting this typo.
You are right, I forgot to update some of the old `jb`.
I just fixed the manuscript and I'll publish a corrected edition in the next few days.

Regards,
Benoit

liuzheng1990, 2019-03-15:
No problem. To be honest I haven't done my purchasing yet. I was going to buy it today because it looks very useful to me. After you publish your corrected edition, I can't wait to place my order!

bblanchon, 2019-03-17:
The new revision is online ;-)

closed 2019-03-17 by bblanchon