bblanchon/ArduinoJson #932: No Error but empty(?) JsonDoc - MQTT - ESP8266

Skalv, 2019-03-15:
Hi,
First, I'm a beginner in C++, maybe mistake come from here...

I use ArduinoJSON for deserialize JSON from MQTT payload.
My code :
```
void mqtt_callback(char* topic, byte* payload, unsigned int length) {
  int i = 0;
  char s[length];

  for(i=0; i<length; i++) {
    s[i] = (char)payload[i];
  }
  s[i] = '\0';

  // char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
  DynamicJsonDocument jsonDoc(256);
  DeserializationError error = deserializeJson(jsonDoc, s);
  //JsonObject root = jsonDoc.as<JsonObject>();

  Serial.print("test1 :");
  Serial.println(jsonDoc["sensor"].as<char*>());
  Serial.print("error :");
  Serial.println(error.c_str());

  const char* sensor = jsonDoc["sensor"];
  Serial.print("sensor : ");
  Serial.print(sensor);
  Serial.println(" .");
}
```

And the Serial return : 
```
test1 :
error :Ok
sensor :  .
```

If I use JSON string likethe commented one (not from payload) code work...

Any ideas ? Thanks.

bblanchon, 2019-03-17:
Hi @Skalv,

You don't need to duplicate `payload` to `s`; instead, you can pass `payload` and  `length` directly to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/):

```c++
DeserializationError error = deserializeJson(jsonDoc, payload, length);
```

However, it doesn't explain your issue...

Can you check the content of `payload`?
Can you call [serializeJsonPretty()](https://arduinojson.org/v6/api/json/serializejsonpretty/) to see how the document looks?

Regards,
Benoit

bblanchon, 2019-03-17:
I added the following entry in the FAQ:

* [How to use ArduinoJson with PubSubClient, the MQTT library?](https://arduinojson.org/v6/faq/how-to-use-arduinojson-with-pubsubclient/)

Let me know if you have any comment.

Skalv, 2019-03-18:
Hi,
The payload send is : 
```"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}"```

I tried deirectly with payload on deserializeJson 
```
  StaticJsonDocument<256> jsonDoc;
  DeserializationError error = deserializeJson(jsonDoc, payload, length);
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
  Serial.print("error :");
  Serial.println(error.c_str());

  Serial.print("Pretty JSON : ");
  serializeJsonPretty(jsonDoc, Serial);
  Serial.println();

  Serial.print("test1 :");
  Serial.println(jsonDoc["sensor"].as<char*>());
  
  const char* sensor = jsonDoc["sensor"];
  Serial.print("sensor : ");
  Serial.print(sensor);
```

But same result :/
```
=== NEW MESSAGE ===
Topic: test
Length: 72
error :Ok
Pretty JSON : "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}"
test1 :
sensor : 
```

Thanks for the FAQ, that was missing ;)

bblanchon, 2019-03-19:
Hi @Skalv,

It seems that the quotes are escaped, so the whole document is a *string*, not an *object*.

In other words, I think the JSON input is the following *string*:

```json
"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}"
```

whereas it should be the following *object*:

```json
{"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}
```

Indeed, the quotes are escaped in the sample program, because C++ requires it, but they shouldn't be escaped in the actual JSON input.

Regards,
Benoit

MoFanFr, 2019-03-21:
I have exactly the same problem.
Code
```
void LoadingDataFromSPIFFS()
{
  if (SPIFFS.exists("/config.json")) {
    //file exists, reading and loading
    Serial.println("reading config file");
    fs::File configFile = SPIFFS.open("/config.json", "r");
    if (configFile) {
      Serial.println("opened config file");
      size_t size = configFile.size();
      // Allocate a buffer to store contents of the file.
      char buf [size+1];
      uint8_t i =0;

      while((configFile.available()))
      {
        buf[i]=configFile.read();
        i++;
      }
      buf[i]='\0';
      Serial.println(buf);
      DynamicJsonDocument doc(256);

      //char input[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
      DeserializationError error = deserializeJson(doc, buf, strlen(buf));
      if (error)
      {
        Serial.print(F("deserializeJson() failed with code "));
        Serial.println(error.c_str());
        return;
      }
      Serial.println(error.c_str());
      JsonObject json = doc.to<JsonObject>();
      //serializeJson(doc, Serial);

       
      Serial.printf(doc["mqtt_server"] | "not found\n");
  	  if (json.isNull())
      {
        Serial.println("parsed json");

      }
      else Serial.println("error in json");
    }
  }
}
```
Serial output

> reading config file
> opened config file
> {"mqtt_server":"192.168.0.57","mqtt_port":"1883","blynk_token":"YOUR_BLYNK_TOKEN","ip":"192.168.0.227","gateway":"192.168.0.1","subnet":"255.255.255.0","SSID":"xxxxx","psk":"yyyyy"}
> Ok
> not found
> error in json

The json in the serial output is the variable "buf".  I have checked the json on the website https://jsonlint.com/ => no error.
I have tried with the json of the variable input (in comment) but it doesn't work too.

I miss probably something...

Skalv, 2019-03-22:
@bblanchon is works !!
All this time lost for a quotes...

Thank you !

@MoFanFr I can't help you but personnaly I use [this library](https://github.com/SergiuToporjinschi/settingsmanager) for my config file, and it works perfectly !

BR,
Florent

MoFanFr, 2019-03-22:
Hello, 
can you tell me where was the missing quotes ? maybe this will help me too.
BR,
Benoit

bblanchon, 2019-03-23:
Hi @MoFanFr,

```c++
JsonObject json = doc.to<JsonObject>();
```

The line above clears the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) and creates an object as its root.
Instead, you need to write:

```c++
JsonObject json = doc.as<JsonObject>();
```

See also:

* [`JsonDocument::as<T>()`](https://arduinojson.org/v6/api/jsondocument/as/)
* [`JsonDocument::to<T>()`](https://arduinojson.org/v6/api/jsondocument/to/)
* [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/)

Regards,
Benoit

MoFanFr, 2019-03-25:
Thank you so much for your help !
I lose a lot of time for this error.

bblanchon, 2019-03-25:
You're welcome, thank you for using ArduinoJson

closed 2019-03-25 by bblanchon