bblanchon/ArduinoJson #934: Passing string to key

wamburu, 2019-03-17:
Am trying to pass as string value to a key, I can do this with directly with v5 but returns null with v6.
Here is the snippet: 

`const size_t capacity = JSON_OBJECT_SIZE(3);`
`DynamicJsonBuffer jsonBuffer(capacity);`

`JsonObject& root = jsonBuffer.createObject();`
`String name_ = "Sensor 1";`
`root["value"] = "10";`
`root["name"] = name_;`

`root.printTo(Serial);`

From this https://github.com/bblanchon/ArduinoJson/issues/258#issuecomment-205877000 `jsonBuffer.strdup(someStringVariable);` seems to pass the string but it is obsolete in v6. 
Anything am missing?

bblanchon, 2019-03-18:
Hi,

You can use a `String` as a key or a value:

```c++
String key = "key";
String value = "value";
doc[key] = value;
```

However, you must increase the capacity of the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) because it makes a copy of all the `String` instances.

Regards,
Benoit

wamburu, 2019-03-18:
That fixes the issue, thanks.

closed 2019-03-18 by wamburu

bblanchon, 2019-03-19:
You're welcome.
Thank you for using ArduinoJson!