bblanchon/ArduinoJson #935: How to use it from gcc on RPI

kuzmanzs, 2019-03-19:
Hi,

I would like to use your code on RPI.
I have downloaded the header and zip from release.
Where should I put the zip? Should I unpack it?

I would like to use it from command line with gcc in a simple hello word json.

Br,
Zsolt

kuzmanzs, 2019-03-20:
I have runned cmake. with original (repo version) CMakeLists.txt
then make
..
Scanning dependencies of target msgpack_fuzzer
[ 98%] Building CXX object fuzzing/CMakeFiles/msgpack_fuzzer.dir/msgpack_fuzzer.cpp.o
[ 99%] Building CXX object fuzzing/CMakeFiles/msgpack_fuzzer.dir/fuzzer_main.cpp.o
[100%] Linking CXX executable msgpack_fuzzer
[100%] Built target msgpack_fuzzer

I could not find novhere the compilled/installed lib.
I use the lib from Arduino IDE but I never tried to ise it from gcc.

bblanchon, 2019-03-20:
Hi @kuzmanzs,

ArduinoJson is a header-only library; there is nothing to install 😉 

Simply download the latest `ArduinoJson-6.x.x.h` from GitHub:
https://github.com/bblanchon/ArduinoJson/releases/latest/

Save it with your project's files, where the compiler can find it.

Regards,
Benoit

kuzmanzs, 2019-03-26:
thanks, it works.

Br,
Zsolt

bblanchon, 2019-03-27:
You're welcome thank you for using ArduinoJson

closed 2019-03-27 by bblanchon