bblanchon/ArduinoJson #939: Do you advise the use of SPIFFS for a LONG json?

clydde, 2019-03-22:
Hello, after trying successfully the deserialization document in "Chunks" of a json hosted on an SD from Arduino One, I decided to try the same with a NodeMCU, the speed has been improved exponentially, and now I am thinking of the idea of ​​" to host the json (120kb) in the SPIFFS memory of the NODEMCU to save me the SD and Card modules, and read the document in Chunks from SPIFFS.

Is the change advisable?
Pros ?, Cons?

Thank you so much!

bblanchon, 2019-03-22:
Hi @clydde,

It should work as expected and requires very few changes to the code you already wrote for SD card.
If parsing in chunks is suitable for you, then yes, it's a good solution.

Regards,
Benoit

clydde, 2019-03-23:
Done
Although it is much slower than with SD. I thought it would be the same or faster. Maybe it's only in the initial load of the sketch, ... I'll continue to investigate

clydde, 2019-03-23:
Hello again, I have done reading tests of my json of 2000 keys.
The time to return the first key of the json hosted on SPIFFS is immediate,
But to return the key 2000 it takes 10 seconds.

The same with SD (return key 2000) takes 1 second.

I do not understand anything, I guess it would be just as fast.

bblanchon, 2019-03-23:
Hi @clydde,

[According to SPIFFS's wiki](https://github.com/pellepl/spiffs/wiki/Performance-and-Optimizing#reading-files) there is a significant overhead when the program read bytes one by one, as ArduinoJson does.

To increase the speed, the best solution is probably to insert a buffering proxy between ArduinoJson and SPIFFS. This class should derive from [Stream](https://www.arduino.cc/reference/en/language/functions/communication/stream/) and pull chunks from the SPIFFS.

Something along those lines:

```c++
BufferedStream bufferedStream(file);
deserializeJson(bufferedStream);
````

I currently don't have the time to write the `BufferedStream` class, but I can do it later if you need.

Regards,
Benoit

clydde, 2019-03-23:
Hi @bblanchon ,

It would be great to have the Class that comments.

Thank you very much

clydde, 2019-03-26:
I managed to lower the search for 4 seconds, changing from 80MHz to 160MHz in the CPU,  but 6 seconds are still too long.

I have been watching the stream class but I do not have enough knowledge to apply it.

I would be very grateful if you could help me in this matter.

Thank you so much!

bblanchon, 2019-03-31:
Hi @clydde,

Please apologize for the delay.
I realized that this class deserved to be tested properly, so I created a project for it.

Please check out https://github.com/bblanchon/StreamUtils and include `StreamUtils.h`.
Then before your read loop add:

```c++
auto bufferedFile = bufferizeInput(file);
```

and replace the call to deserializeJson() with 

```c++
deserializeJson(doc, bufferedFile);
```

This will read the file by chunks of 64 bytes, which should greatly improve the performance.

Regards,
Benoit

clydde, 2019-04-01:
Many, thank you very much, 
I will try it!

clydde, 2019-04-01:
Hello the compiler return an error and impossible try  the buffered,

The ino code its similar to this: 

`#include <ArduinoJson.h>
#include <StreamUtils.h>

void setup() {
SPIFFS.begin();
  Serial.println("Inicialitzant SPIFFS ...");
  if (!SPIFFS.begin()) {
    Serial.println("Error SPIFFS!");
    while (1);
  }
}
void loop() {
}
void myFuncion(String fromTar){
 //File myFile = FS.open("users.txt");// working on SD
  
  File myFile = SPIFFS.open("/users.txt", "r"); // Wordking on SPIFFS
  
  // New buffered add
  auto bufferedStream = bufferizeInput(myFile); // TEST SPIFFS BUFFERED

  if (!myFile.find("\"usr\":")) {
    Serial.println(F("Array \"usr\" not allowed"));
  }

  DynamicJsonDocument doc(102);

  myFile.find("[");

   
  do {
    // deserializeJson(doc, myFile); // working  on SD
    //auto err = deserializeJson(doc, myFile); // working SPIFFS
    
    auto err = deserializeJson(doc, bufferedStream); // TEST SPIFFS BUFFERED
    
    if (err) {
      Serial.print(F("deserializeJson() failed with code "));
      Serial.println(err.c_str());
    }
    .....
    .....
    .....
    }

  } while (myFile.findUntil(",", "]"));

  Serial.println("END Json.");
  myFile.close();
}
}
`
And the compiler says:

`n file included from C:\Users\webmaster\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:7:0,

                 from C:\Users\webmaster\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:33,

                 from C:\Users\webmaster\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\webmaster\Documents\Arduino\CSMS\nodeMCUcSMS_3\nodeMCUcSMS_3.ino:9:

c:\users\webmaster\documents\arduino\libraries\arduinojson\src\arduinojson\deserialization\deserialize.hpp: In instantiation of 'ArduinoJson6100_000::DeserializationError ArduinoJson6100_000::deserialize(ArduinoJson6100_000::JsonDocument&, TStream&, ArduinoJson6100_000::NestingLimit) [with TDeserializer = ArduinoJson6100_000::JsonDeserializer; TStream = StreamUtils::BufferedStream<fs::File, 102u>]':

C:\Users\webmaster\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:413:64:   required from 'ArduinoJson6100_000::DeserializationError ArduinoJson6100_000::deserializeJson(ArduinoJson6100_000::JsonDocument&, TInput&, ArduinoJson6100_000::NestingLimit) [with TInput = StreamUtils::BufferedStream<fs::File, 102u>]'

C:\Users\webmaster\Documents\Arduino\CSMS\nodeMCUcSMS_3\nodeMCUcSMS_3.ino:105:51:   required from here

c:\users\webmaster\documents\arduino\libraries\arduinojson\src\arduinojson\deserialization\deserialize.hpp:72:48: error: 'Stream' is an inaccessible base of 'StreamUtils::BufferedStream<fs::File, 102u>'

              doc.memoryPool(), makeReader(input),

                                                ^

exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).

`

Could you help me find the problem?

Thank you very much (again)

clydde, 2019-04-01:
I locate the pluh-in in

MyProjectFolders / libraries / ArduinoJson /src/
StreamUtils.h
StreamUtils (Folder) / BufferedStream.hpp

bblanchon, 2019-04-02:
I just updated the code, please give it another try.

clydde, 2019-04-02:
Hi, now the compiler works, but the NOdeMCU Crash, The Serial Motinor return:

`ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset` 

Analyzing the result:

`>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffc0 offset: 01a0
3fffffa0:  3fffdad0 00000000 3ffee600 40203418  
3fffffb0:  feefeffe feefeffe 3ffe84fc 401009ed  
<<<stack<<<`

`ecoding stack results
0x40203418: loop_wrapper() at C:\Users\webmaster\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 125`

bblanchon, 2019-04-05:
I pushed a new revision, please try again.

clydde, 2019-04-05:
Hi,

Missing little thing!

Now I do not experience any errors, but ...

If I look for the first JSON KEY, everything works fine, but if I look for any other JSON KEY the result is a controlled error:

deserializeJson() failed with code InvalidInput
deserializeJson() failed with code InvalidInput
deserializeJson() failed with code InvalidInput
...

I leave here the code I am using to try it:

`void searchTar (String enterString) {

File myFile = SPIFFS.open("/users.txt", "r"); 

auto bufferedStream = bufferizeInput(myFile); // TEST SPIFFS BUFFERED

  if (!myFile.find("\"usr\":")) {
    Serial.println(F("Array \"usr\" no find!"));
  }

  DynamicJsonDocument doc(102);

  myFile.find("[");

  do {
   
    auto err = deserializeJson(doc, bufferedStream);

    if (err) {
      Serial.print(F("deserializeJson() failed with code "));
      Serial.println(err.c_str());
      break;
    }

    String theFind = doc["tar"].as<char*>();

    if (latar == enterString) {
      Serial.println(doc["pin"].as<char*>());
      break;
    }
    

  } while (myFile.findUntil(",", "]"));

  myFile.close();
  
}`

The json's pattern is this:

`
	"usr": [{
		"tar": "f11hf0876fg0h2871n657465er4476h1",
		"pin": "8wKqhX/Q3d/8ECyGPc8bDA=="
	}]
}`

And I changed its value by 64 for 102 in the file BufferedStream.hpp

Thankyou!

bblanchon, 2019-04-06:
Hi @clydde,

The program should now use only `bufferedStream`, but it still uses `myFile` in several places:

* `myFile.find("\"usr\":")`
* `myFile.find("[");`
* `myFile.findUntil(",", "]")`

Replacing these lines should fix the issue.

Regards,
Benoit

clydde, 2019-04-06:
upsss! WORK!!! and not only work, it does it just as fast as with SD!!!!!

Many thanks for the addition and for the patience!

bblanchon, 2019-04-06:
You're welcome :-)
I'll polish this library and publish it so that anyone can use it.
I guess I can close this issue then.

closed 2019-04-06 by bblanchon