bblanchon/ArduinoJson #940: What is the Benefit of V6 compared to V5

hasenradball, 2019-03-22:
Hi Benoit,

can you please tell me in a view word the benefit of upgrading to V6 when actually using V5 of ArduiniJSON?

Thanks

bblanchon, 2019-03-22:
Hi @hasenradball,

In short:

- simpler to use (no reference and JsonDocument)
- tells why parsing fails
- MessagePack
- lower memory usage

But if your code works fine with v5, keep it like this.

Regards,
Benoit

hasenradball, 2019-03-23:
Hi Benoit,

Thank you very much.
Is the V6 documented in a new version of the book?

Thanks

closed 2019-03-23 by hasenradball

bblanchon, 2019-03-23:
Yes, the book is in sync with 6.9.

hasenradball, 2019-03-24:
One last question Benoit,

if I have one book which has original foil and is in original wrapping.
Is ther the posibility to excahange this to a versin which has the V6.9 inside?

Regards Frank

bblanchon, 2019-03-24:
Please send me an email, we'll fix that.