bblanchon/ArduinoJson #941: using reference in ArduinoJson/Document/BasicJsonDocument.hpp ?

fsicre, 2019-03-23:
```
  BasicJsonDocument& operator=(const BasicJsonDocument& src) {
    // reallocPoolIfTooSmall(src.memoryUsage()); BUG ?
    reallocPoolIfTooSmall(src -> memoryUsage());
    set(src);
    return *this;
  }

  template <typename T>
  BasicJsonDocument& operator=(const T& src) {
    // reallocPoolIfTooSmall(src.memoryUsage()); BUG ?
    reallocPoolIfTooSmall(src -> memoryUsage());
    set(src);
    return *this;
  }
```

bblanchon, 2019-03-24:
Hi @fsicre,

I'm sorry, I don't understand.
Can you explain the problem?

Regards,
Benoit

fsicre, 2019-03-24:
const T& src

src is a reference 

you got to de-reference it with a ->, no ?

reallocPoolIfTooSmall(src.memoryUsage());  <= BUG

reallocPoolIfTooSmall(src -> memoryUsage()); <= FIX

seen too times in file BasicJsonDocument.hpp

bblanchon, 2019-03-31:
Hi @fsicre,

I'm still don't understand.
C++ references are dereferenced using the dot syntax if that's your question.
Can you tell me what you expect from me?
(Feel free to answer in French if you prefer).

Regards,
Benoit

closed 2019-04-19 by bblanchon

fsicre, 2019-04-20:
La dernière fois que j'ai compilé un projet incluant ArduinoJson, le compilateur a refusé
en modifiant les lignes
    reallocPoolIfTooSmall(src.memoryUsage());
en
    reallocPoolIfTooSmall(src -> memoryUsage());
la compilation est passée.

Désolé pour mon manque de clarté.
Je comprends presque rien au C++.
Mais gcc oui.

bblanchon, 2019-04-20:
Est-ce que tu peux coller la sortie complete du compilateur STP?