bblanchon/ArduinoJson #943: Object Serialization With .as<String>()

abaskin, 2019-03-24:
I've noticed that calling `.as<String>()` on a JsonObject returns the object as a minified JSON document. The behavior also works with std::string, but not with char*. This is very convenient when you want to only serialize a portion of the document. My question is if this behavior is full supported (i.e. not a side effect that may go away in the future) and is there any downside to using this approach as opposed to calling` serializeJson()`?

bblanchon, 2019-03-24:
Hi @abaskin,

I added this feature on purpose, and I plan to keep it.

The downsides are:

- only works with `String` and `std::string`
- only produces minified JSON

To be clear, you can also call `serializeJson()` to serialize a portion of the document:

```c++
serializeJson(doc["config"]["wifi"], output);
```

Regards,
Benoit

closed 2019-04-06 by bblanchon

mikee47, 2019-04-30:
I note this behaviour change from V5 to V6. Essentially, the rule seems to be that when converting from `json[key].asString()` in V5, use `json[key].as<const char*>()` to get the same behaviour. If that's correct, why was `asString()` removed?