bblanchon/ArduinoJson #946: big JSON file of web radio streaming site

loszelos, 2019-03-25:
Hello, 

I do graduading thesis web streaming radio with MEGA2560 and ESP32 with VS1053.
I thought lot of time, how to get actual list of stations and I found this super page http://www.radio-browser.info/webservice . There are JSONs files , but they are sooo big. 
I wanted to ask, if is possible load to array, and by button jump to next station (block of JSON) and show on TFT display. I thought , to make some menu and submenus , and after that selected JSON block, where is the link for play and send it to ESP32 via I2C. For knowing, I need MEGA there for next modules and etc. 
I don't problem to purchase your book, if is possible to find answer there.

Thanks for any help and answers !!

bblanchon, 2019-03-26:
Hi @loszelos,

Please read:

* [How to deserialize a very large document?](https://arduinojson.org/v6/faq/how-to-deserialize-a-very-large-document/)

Regards,
Benoit

loszelos, 2019-03-26:
> Komu: bblanchon/ArduinoJson <ArduinoJson@noreply.github.com>
> site (#946)
> "

Thanks for your kind mail.

I purchased your book today. However, I have problem how to turn this piece 
of code for aplying for read from SD card . Is it possible to write me back 
some example? 

//////////////////////////////////////////////

client.find(beginingOfList);

Chapter 7 Case Studies 237
do {
// Parse the JSON object
StaticJsonDocument<1024> doc;
deserializeJson(doc, client);
// Display the values
Serial.println(doc["dt"].as<long>()); // timestamp
Serial.print(doc["main"]["temp"].as<float>()); // temperature
Serial.print(doc["weather"][0]["description"].as<char*>()); // text
// Skip the comma or stop the loop
} while (!client.findUntil(comma, endOfArray));
// Disconnect from host
client.stop();

/////////////////////////////////////////////////

I used standard SD library. 

Thanks for your kind help! 

---------- Původní e-mail ----------

   (https://arduinojson.org/v6/faq/how-to-deserialize-a-very-large-document/)
   

(https://github.com/bblanchon/ArduinoJson/issues/946#issuecomment-476506253)
"

loszelos, 2019-03-26:
Thanks for your kind mail.
I purchased your book today. However, I have problem how to turn this piece of code for aplying for read from SD card . Is it possible to write me back some example? I think example for OpenWeatherMap on mkr1000. There is only for web but I would like to modify for SD card. Thanks !

bblanchon, 2019-03-27:
Hi @loszelos,

Thank you for purchasing the book!
The book contains an example that reads from the SD card: section 3.10.1 and folder `ReadFromSdCard`.
You can also check the example [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/); it uses the SD card too.

Regards,
Benoit

loszelos, 2019-03-31:
Hi, I tried to parse code from http://www.radio-browser.info/webservice/
json/countries but I'm not able to do that. 

I have arduino uno with ethernet shield and I cannot connect to the server. 
I don't know , where is the wrong, is it possible to help me? 

I have there this: 

 EthernetClient client;

  if (!client.connect("www.radio-browser.info", 80)) {
    Serial.println(F("Connection failed"));
    return;
  }

but returned on serial Connection failed.

loszelos, 2019-03-31:
I'm so sorry, just now I am connected. My wrong . But I tried whole 
afternoon parsing code and I'm still fail. 

Is it possible to help me exactly with sketch for radio-browser.info? 

Thanks ;-)

loszelos, 2019-03-31:
> I tried whole afternoon parsing code and I'm still totally fail in JSON at all. Is it possible to help me for example exactly with sketch for radio-browser.info? for example this : [{"name":"AAC","value":"AAC","stationcount":"863"},{"name":"AAC+","value":"AAC+","stationcount":"4301"},{"name":"AAC+,H.264","value":"AAC+,H.264","stationcount":"2"},{"name":"AAC,H.264","value":"AAC,H.264","stationcount":"159"},{"name":"FLV","value":"FLV","stationcount":"5"},{"name":"MP3","value":"MP3","stationcount":"15159"},{"name":"MP3,H.264","value":"MP3,H.264","stationcount":"2"},{"name":"OGG","value":"OGG","stationcount":"245"},{"name":"UNKNOWN","value":"UNKNOWN","stationcount":"1952"},{"name":"UNKNOWN,H.264","value":"UNKNOWN,H.264","stationcount":"9"}] I would like to use method for big files, input.find("\"characters\":["); do { deserializeJson(doc, input); // ... } while (input.findUntil(",","]")); Also I think, that it's great example for your book, when you will update it ;-) Thanks ;-)

bblanchon, 2019-04-06:
Hi @loszelos,

Indeed, this radio-browser API would be another great example for the book.

It think the only mistake is that `find()` looks for the string `"characters":[` which is not present in the input. You need to change to:

```c++
input.find("[");
do {
    deserializeJson(doc, input);
    // ...
} while (input.findUntil(",","]"));
```

Also, you may need to enable Unicode escape sequence with [ARDUINOJSON_DECODE_UNICODE](https://arduinojson.org/v6/api/config/decode_unicode/)

Regards,
Benoit

closed 2019-04-19 by bblanchon