bblanchon/ArduinoJson #947: ArduinoJson v6 not compiling on Sloeber Plugin

karpenet, 2019-03-25:
Using ArduinoJSON v6 produces compilation error. Suggests that StaticJsonDocument is an invalid keyword. But using keyword StaticJsonBuffer, the code compiles without any hassle.

While the Arduino IDE displays V6.10 as the currently installed version of ArduinoJSON, the Sloeber plugin shows v5 versions in the preferences menu.

bblanchon, 2019-03-26:
Hi @karpenet,

I'm sorry, I'm not familiar with the Sloeber plugin.
I suppose it looks for ArduinoJson in the wrong location, possibly in an older installation of Arduino.
You can look for `ArduinoJson.h` on your hard-drive to see where the library is installed.
Then, maybe you should ask for assistance to the Sloeber team.

Regards,
Benoit

bblanchon, 2019-04-06:
Did you make it work?

closed 2019-04-19 by bblanchon