bblanchon/ArduinoJson #950: Why must I create a separate config object - why can't I use the json object directly?

Daemach, 2019-03-26:
The examples show parsing a json file then using the parsed data to create a new config object. 
 Why can't I just use a static buffer (on the stack), parse my config file into it and then use that object directly throughout my code rather than using it to create another config object?  This seems like the perfect solution - I could change values directly in the json object and save it whenever changes are made without converting to and from a separate config object.

bblanchon, 2019-03-31:
Hi @Daemach,

Here are three reasons, but you can probably find more:

1. [JsonDocument](https://arduinojson.org/v6/api/jsondocument/) contains a monotonic allocator (i.e., it cannot release memory), so it will be full after a few iterations. You can see that as a memory leak in the [JsonDocument](https://arduinojson.org/v6/api/jsondocument/), but it's [by-design](https://arduinojson.org/2018/12/07/version-6-7-0/).
2. Overhead: passing through the JSON object model is less efficient regarding:
   - memory usage,
   - execution speed,
   - code size.
3. Type safety: once your data is captured in C++ structures, you can rely on the security provided by the type system; you don't have to check that such member is present and have the right type.

@Daemach, that was a good question, and I thought arduinojson.org already answered it. I'll be very grateful if you could advise me where I could add this answer to the site so that other users can find it.

Regards,
Benoit

Daemach, 2019-04-01:
I would add it directly to the config file example ino file.  I always go
straight to the code to see what will and will not work for me.

Thank you for answering my questions and for all of your hard work!

bblanchon, 2019-04-01:
Got it! Thank you for the feedback, I'll add that sometime soon.

closed 2019-04-01 by bblanchon