bblanchon/ArduinoJson #951: Error in Deserializing but output okay

bill-orange, 2019-03-26:
I am using the latest Version 6 of ArduinoJson on an ESP8266.  I am deserializing the 3 day forecast from OpenWeatherMap.

It is not small.  In DynamicJsonDocument doc(capacity), capacity is around 26000.  I get the payload with `json = http.getString().c_str(); `and deserialize with `deserializeJson(doc, json);` .  It appears to work.  I only need the first two arrays and they come through fine.  However, I am getting the error: `deserializeJson() failed: IncompleteInput`.  Although, these functions in my sketch give a usable result,  I am reluctant to get forward with an error.   Increasing capacity does not seem to make any difference.

Any suggestions on how to go forward without leaving myself open to some weird error or result later down the road?  While still declaring the HTTPClient object http, I tried this: `deserializeJson(doc, http.getString().c_str())`  I got the same IncompleteInput error but without any output,

I am at the ragged edge of my understanding or a bit beyond the edge so any direction would be greatly appreciated.  I suppose going forward with what I have is a possibility but it seems wrong.

bblanchon, 2019-03-27:
Hi @bill-orange,

Often, [`IncompleteInput`](https://arduinojson.org/v6/api/misc/deserializationerror/) is due to a timeout; you can try to call [`setTimeout()`](https://www.arduino.cc/en/Reference/StreamSetTimeout) to set a higher value.
Also, you could try to print the content of the input to the Serial to check if the JSON document is complete.

Regards,
Benoit

bill-orange, 2019-03-27:
@bblanchon 

Thank you for the prompt reply and, of course, the great software.

  I presume that setTimeout() would apply to http, so I added the line  http.setTimeout(10000); right after the http class declaration.  No change.  I added a print statement for json.  It looked okay so I cut and pasted it into json assistant.  It worked just fine.  Adding another 5000 to capacity also had no effect.  I also pasted the string in a json parser web site.  It parsed normally.  Nesting appears to be 5 deep.  Is that okay?

I am leaning toward living with the error.  

Your thoughts?

bblanchon, 2019-03-28:
Hi @bill-orange,

In one of my projects, it had intermittent `IncompleteInput` like this.
Like you, it was on ESP8266 and only with very large inputs.

To track the bug, I implemented the logging proxy as described in [Mastering ArduinoJson](https://arduinojson.org/book/). The idea is to insert a class between the `WiFiClientSecure` and `deserializeJson()` this class behaves like the `WiFiClientSecure` excepts that it also prints the content to the serial.

From what I saw, the input was truncated, and I concluded that there must be a race condition in the `WiFiClientSecure`, or any other problem that is beyond my zone of influence.
In the end, I decided to live with the error.

Regards,
Benoit

bill-orange, 2019-03-28:
@bblanchon Thanks for your thought,  I think I will also live with the error report.  My code has been running for three days now without any issues.  Memory is consistenly: `freeMemory()= 33576`  There does not appear to be a shortage of memory or a memory leak.  The race condition certainly sounds like a likely cause.  I tried changing lwIP variants.  It made no difference.  I tried change the clock from 80 mHz to 160 mHz that also made no difference (except for occasionally crashing SPIFFS memory).

Thanks again.

bblanchon, 2019-03-29:
You're welcome. Thank you for sharing this.

closed 2019-03-29 by bblanchon