bblanchon/ArduinoJson #952: deserializeJson results in InvalidInput

naitsimp, 2019-03-27:
I am trying to switch from v5 to v6, serializeJson is working as expected, but I have troubles with the deserializeJson.
When a MQTT message arrives the function callback is called in which I call the function processReceivedJson to process the incoming data.
I always get an **InvalidInput** error.
The Json which arrives is: `{"state":"ON"}`
Before StaticJsonDocument I publish a message and get the correct message back: `{"aaa":"{\"state\":\"ON\"}"}`
Testing for DeserializationError check results in: `{"error":"InvalidInput"}`
and the message gets weird, I get `{"bbb":"state"}`

Can someone help me what I am doing wrong? Struggling for hours now.

When I publish {"22":"ON"} I pass the DeserializationError check and get {"OK":"OK"} back.

My functions:
```
void callback(char* topic, byte * payload, unsigned int length) {
  debug_print("Message arrived: " + String(topic));

  char message[length + 1];            
  for (int i = 0; i < length; i++) {   
    message[i] = (char)payload[i];
  }
  message[length] = '\0';

  processReceivedJson(message, String(topic));  
}

bool processReceivedJson(char* message, String topic) {         

  Payload[0][0] = "aaa"; Payload[0][1] = String(message);
  sendState("error", Payload, 1, 0);

  StaticJsonDocument<BUFFER_SIZE> doc;          
  deserializeJson(doc, message);

  //Test if parsing suceeds
  DeserializationError error = deserializeJson(doc, message);
  if (error) {
    Payload[0][0] = "error"; Payload[0][1] = String(error.c_str());
    sendState("error", Payload, 1, 0);
    Payload[0][0] = "bbb"; Payload[0][1] = String(message);
    sendState("error", Payload, 1, 0);

    Serial.print(F("deserializeJson() failed with code "));
    Serial.println(error.c_str());
    //return;
  }
  else {
    Payload[0][0] = "OK"; Payload[0][1] = String("OK");
    sendState("error", Payload, 1, 0);
  }

  if (topic == subscribe_topic_led_light_1_cmnd) {
    if (doc["state"]) {          
      if (doc["state"] == on_cmd) {       //on_cmd == "ON"  
        
      CODE FOR TURNING LIGHT ON 

      }
      else if (doc["state"] == off_cmd) {      //off_cmd == "OFF"  

      CODE FOR TURNING LIGHT OFF
 
      }
    }
  }
```

bblanchon, 2019-03-27:
Hi @naitsimp,

Please print the content of the input to the Serial port so we can verify that it's valid.
Most of the time, [InvalidInput](https://arduinojson.org/v6/api/misc/deserializationerror/) is caused by an input that contains escaped strings; see #932 for example.

Also, have a look at:

* [How to use ArduinoJson with PubSubClient, the MQTT library?](https://arduinojson.org/v6/faq/how-to-use-arduinojson-with-pubsubclient/)

Regards,
Benoit

naitsimp, 2019-03-27:
OK, debugging with Serial Monitor now.
Added a few Serial.print.
New Code:
```
void callback(char* topic, byte* payload, unsigned int length) {
  debug_print("Message arrived: " + String(topic));

  char message[length + 1];            //create an array named message with the size of the mqtt message + to add the 0
  for (int i = 0; i < length; i++) {   //prints each byte of the message, prints the payload
    message[i] = (char)payload[i];
  }
  message[length] = '\0';
  debug_println(message);

  StaticJsonDocument<256> doc0;
  deserializeJson(doc0, payload, length);
  const char* sensorA = doc0["state"];
  Serial.print("0 sensorA:\t|-"); Serial.print(sensorA); Serial.println("-|\n");

  Serial.print("1 topic:\t|-");  Serial.print(topic); Serial.println("-|");
  Serial.print("1 message:\t|-"); Serial.print(message); Serial.println("-|\n");

  processReceivedJson(message, String(topic));  //process the message which arrived, and decide what to do
}

/********************************** START PROCESS JSON *****************************************/
bool processReceivedJson(char* message, String topic) {         //decides what to do after the mqtt message arrived

  Serial.print("2 topic:\t|-"); Serial.print(topic); Serial.println("-|");
  Serial.print("2 messsage:\t|-"); Serial.print(message); Serial.println("-|\n");

  StaticJsonDocument<BUFFER_SIZE> doc;           // Memory pool for JSON object tree
  deserializeJson(doc, message);

  Serial.print("3 messsage:\t|-"); Serial.print(message); Serial.println("-|");
  const char* sensorB = doc["state"];
  Serial.print("3 sensorB:\t|-"); Serial.print(sensorB); Serial.println("-|\n");

  //Test if parsing suceeds
  DeserializationError error = deserializeJson(doc, message);
  if (error) {

    Serial.print("4 messsage:\t|-"); Serial.print(message); Serial.println("-|");
    //Serial.print(doc["state"]); //error: call of overloaded 'print(ArduinoJson6100_000::enable_if<true, ArduinoJson6100_000::MemberProxy<ArduinoJson6100_000::JsonDocument&, const char*> >::type)' is ambiguous
    Serial.print(F("deserializeJson() failed with code: "));
    Serial.println(error.c_str());
    //return;
  }
  else {
    Payload[0][0] = "OK"; Payload[0][1] = String("OK");
    sendState("error", Payload, 1, 0);
  }

  Serial.print("\n5 messsage:\t|-"); Serial.print(message); Serial.println("-|");
  const char* sensorC = doc["state"];
  Serial.print("5 sensorC:\t|-"); Serial.print(sensorC); Serial.println("-|\n");

..... below same code as first post
```

Serial Output:
```
0 sensorA:	|-OFF-|

1 topic:	|-home/kitchen_node_1/led_light_1/cmnd-|
1 message:	|-{"state":"OFF"}-|

2 topic:	|-home/kitchen_node_1/led_light_1/cmnd-|
2 messsage:	|-{"state":"OFF"}-|

3 messsage:	|-state-|
3 sensorB:	|-OFF-|

4 messsage:	|-state-|
deserializeJson() failed with code: InvalidInput

5 messsage:	|-state-|
5 sensorC:	|--|
```

Until 3 everything is as expected.
At 3:
Why does the variable `message` change to only state?
sensorB = OFF, as expected
It does work, but why does `DeserializationError` result in an error?

After that I suspected that the `DeserializationError` check is the reason for the problem. 
I was right, when I comment out the whole if else, everything works as espected. 
Why and what is happening here?

Is it necessary to save the json in a variable like: `const char* sensorA = doc0["state"];`
Is it possible to acces it without saving it in a variable?
`Serial.println(doc["state"]);` not working.
`Serial.print(sensorC);` working.

bblanchon, 2019-03-28:
Hi @naitsimp,

Because the type of the input is `char*`, ArduinoJson uses the zero-copy mode and therefore modifies the input buffer.

Please read:

* [Why is the input modified?](https://arduinojson.org/v6/faq/why-is-the-input-modified/)

Regards,
Benoit

naitsimp, 2019-03-28:
Thank you very much, I was not realizing that I was calling `deserializeJson(doc, message);` twice, first with `deserializeJson(doc, message);` and then in `DeserializationError error = deserializeJson(doc, message);`

**About my last question:**
I want to check if the Json contains the key `state` and which value it has.
The Documentation says I should avoid `root.containsKey("state")` if possible, which I was using in v5.
Do I need to save the Json in a variable like `const char* keys = root["state"]["other1"]["other2"];` first before I can read it with `doc["state"]; doc["other1"]; doc["other2"]`
A if statement like: `if (doc["state"])` or `if (doc["state"] == on_cmd)` is not working.
I would prefer not to save it in a varibale first, because I don't know which keys I will receive in advance.

bblanchon, 2019-03-29:
Hi @naitsimp,

The documentation recommends avoiding `containsKey()` because it often does two lookups.
Try something like that:

```c++
JsonObject state = doc["state"];
if (!state.isNull()) {
  const char* keys = state["other1"]["other2"];

}
```

Regards,
Benoit

naitsimp, 2019-03-29:
Hi, Thank you, sorry to bother you again, but I can't make it work.

My JSON message for a light is `{"state":"ON"}`
As I understand now there is no other way around saving the json in a variable when I want to avoid `containsKey()` in v6.

I am asking again because I do things depending on the received state.
How would I rewrite the if, else if statement below to upgrade to v6 and avoid `containsKey()`?

My v6 Code:
???: What to put here to decide what to do?
```
  StaticJsonDocument<BUFFER_SIZE> doc;

  DeserializationError error = deserializeJson(doc, message);
  if (error) {
    Serial.print(F("deserializeJson() failed with code: "));
    Serial.println(error.c_str());
  }

  if (topic == subscribe_topic_led_light_1_cmnd) {
    if (???) {         
      if (???) { 

        DO SOMETHING

      }
      else if (???) {

         DO SOMETHING

    }
  }
```

This was my v5 Code:
```
  StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(message);

  if (topic == subscribe_topic_led_light_1_cmnd) {
    if (root.containsKey("state")) {         
      if (root["state"] == on_cmd) {          

        DO SOMETHING

      }
      else if (root["state"] == off_cmd) {

         DO SOMETHING

      }
    }
  }
```

bblanchon, 2019-03-30:
There is no need to check the presence of "state".
This will do:

```c++
if (topic == subscribe_topic_led_light_1_cmnd) {
  if (doc["state"] == on_cmd)) { 

  }
  else if (doc["state"] == off_cmd)) {
  
  }
}
```

naitsimp, 2019-03-30:
@bblanchon Thank you so much for your help, my project is working now.

bblanchon, 2019-03-31:
You're welcome @naitsimp. Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project 😉

closed 2019-03-31 by bblanchon