bblanchon/ArduinoJson #953: Determining size of json file being read and setup of DynmaicJsonDocument(xx)

mars000, 2019-03-27:
I'm using latest version 6 library.

making my way across to V6, I am reading a json file using ESP32 and WiFiClient but the size of this json file may vary over time. With V5 it was taken care for me. 

Therefore my question is how do I compute my doc size in this case that I don't know what the size is and from what I can tell when I use DynamicJsonDocument doc(capacity) ... I need to put in the capacity number in bytes which I don't know ahead of time ?

clydde, 2019-03-27:
Hi, I hope I can help you, anyone who feels they can correct my words feel free to do so. Let's go there!

The assistant will help you in that:
https://arduinojson.org/v6/assistant/

Keep in mind that you do not need to put all the JSON in the wizard, but the base structure, so that the library will use those memory sections with

DynamicJsonDocument (xxx)

I am using a 150kb file with 4000 keys and I can increase or reduce it without problem using the technique described in the book as extraction for "Chunks".

I hope to shed some light on your doubt.

bblanchon, 2019-03-28:
Hi @mars000,

Please read [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/faq/how-to-determine-the-capacity-of-the-jsondocument/); I just updated it for version 6.

Regards,
Benoit

closed 2019-04-06 by bblanchon