bblanchon/ArduinoJson #954: Problem with large numbers after moving from v5 to v6 (EXPONENTIATION_THRESHOLD)

patman15, 2019-03-28:
Hi!

Since version 5 of the library I received an array of values with timestamps in milliseconds, e.g.
```
  {
    "val": false,
    "ts": 1553802618889
  }
```
In order to avoid usage of long long or int64_t I used the following:
```
time_t ts; /// unix timestamp
ts = value.get<String>("ts").substring(0, 10).toInt();
```
that avoided decoding the full number and then dividing by /1000 since I only need the 32bit Unix timestamp in seconds. 
Since version 6 this does not work anymore, because the string is transformed into scientific notation and setting `ARDUINOJSON_POSITIVE_EXPONENTIATION_THRESHOLD 1e13` corrupts the numbers.
Is there a way to disable the rewriting to scientific notation?

* target platform is ATSAMD21
* compiler model and version gcc version 4.8.3 20140228 (release) [ARM/embedded-4_8-branch revision 208322]

Thanks for the great library!

bblanchon, 2019-03-29:
Hi @patman15,

It worked with ArduinoJson 5, because this version decodes the integers lazily, i.e., only when you call `as<int>()`.
However, it was not possible to keep this behavior in ArduinoJson 6 because it was incompatible with the `deserializeJson() -> serializeMsgPack()` scenario.

Because the value doesn't fit in a 32-bit integer, ArduinoJson switches to floating point storage to avoid integer overflow. This is a new feature of [version 6.10](https://arduinojson.org/2019/03/22/version-6-10-0/).

I think the best solution is to divide the floating point by 1000 and cast it to an integer:

```c++
ts = int(value["ts"].as<double>() / 1000);
```

If that's not acceptable, you will have to switch to a `long long` storage with [ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v6/api/config/use_long_long/), and then do the same dance:

```c++
ts = int(value["ts"].as<long long>() / 1000);
```

However, I don't recommend using this technique because the `JsonDocument` needs to be bigger.

Regards,
Benoit

patman15, 2019-03-29:
Hi @bblanchon,

thanks for the information. I tried to avoid setting the defines to save flash storage but I now enabled it and optimized a few other things. Btw. using <double> used more flash than enabling long long.

closed 2019-03-29 by patman15