bblanchon/ArduinoJson #955: How to create dynamic Nested Object to Array?

Findus76, 2019-03-28:
Lib version: 6.10.0
Compiler: PlatformIO
Target: ESP8266

I can't figure out how to add dynamic Nested Object to Array.
e.g
{"Type" : "Event", "Event list": [
   {"Name": "Voltage L1", "Value": 230},
   {"Name": Current L1", "Value": 1.23},
   {"Name": "Temperature", "Value": 23.3}]}

It can be up to 45 object in the "Event list" array.
[image]

Brg
Findus

bblanchon, 2019-03-30:
Hi Findus,

The compiler says it cannot find `JsonArray::createNestedObject(const char*)` but it found a `JsonArray::createNestedObject()`.
Indeed, [JsonArray::createNestedObject()](https://arduinojson.org/v6/api/jsonarray/createnestedobject/) doesn't take any argument.

Regards,
Benoit

Findus76, 2019-03-31:
Okey
is this correct? It feels strange in the loop with the creation of NestedObject
[image]
Brg Findus

Findus76, 2019-04-01:
Thanks!

haven't tried yet, have got big problems with the hardware. Need to resolder.
But the compiler runs with 0 error now :)

Big thanks for your review.

Brg
Findus

closed 2019-04-01 by Findus76

bblanchon, 2019-04-03:
You're welcome Findus; thank you for using ArduinoJson.
Don't forget to [cast a start](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project 😉