bblanchon/ArduinoJson #956: Error compiling for board NodeMCU 1.0 (ESP-12E Module)

aldoexpert, 2019-03-29:
I have problem with compiling the NodeMcu, Error is
--------------------
In file included from /Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantOperators.hpp:7:0,
                 from /Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:12,
                 from /Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:8,
                 from /Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:9,
                 from /Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:17,
                 from /Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/aldo/Dropbox/CODE/Arduino/project-att/sketch_att/sketch_att.ino:5:
/Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantCasts.hpp: In instantiation of 'ArduinoJson6100_000::VariantCasts<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson6100_000::MemberProxy<ArduinoJson6100_000::JsonDocument&, const char*>]':
/Users/aldo/Dropbox/CODE/Arduino/project-att/sketch_att/sketch_att.ino:185:36:   required from here
/Users/aldo/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Operators/VariantCasts.hpp:16:35: error: invalid conversion from 'ArduinoJson6100_000::VariantAs<char*>::type {aka const char*}' to 'char*' [-fpermissive]
     return impl()->template as<T>();
                                   ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
--------------------

in my project use the libraries
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>
#include <SPI.h>
#include <RFID.h>
#include <LiquidCrystal_I2C.h>
#include <MD5Builder.h>

Please help

bblanchon, 2019-03-30:
Hi @aldoexpert,

On line 185 of `sketch_att.ino`, there is a `char*` that should be changed to a `const char*`.

Please see:

* [Error: invalid conversion from 'const char*' to 'char*' [-fpermissive]](https://arduinojson.org/v6/faq/error-invalid-conversion-from-const-char-to-char/)

Regards,
Benoit

amandarh2807, 2019-04-06:
I have problem with compiling the NodeMcu, Error is

Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Generating function prototypes...
"C:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/include" "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/lwip2/include" "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\AMANDA~1\AppData\Local\Temp\arduino_build_461809/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10802 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266" "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\variants\nodemcu" "-IC:\Users\Amanda Rahmat\Documents\Arduino\libraries\firebase-arduino-master\src" "-IC:\Users\Amanda Rahmat\Documents\Arduino\libraries\ArduinoJson-6.x\src" "-IC:\Users\Amanda Rahmat\Documents\Arduino\libraries\Arduino_JSON\src" "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src" "-IC:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient\src" "C:\Users\AMANDA~1\AppData\Local\Temp\arduino_build_461809\sketch\coba_yutup1.ino.cpp" -o "C:\Users\AMANDA~1\AppData\Local\Temp\arduino_build_461809\preproc\ctags_target_for_gcc_minus_e.cpp"
In file included from C:\Users\Amanda Rahmat\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:30:0,

                 from C:\Users\Amanda Rahmat\Documents\Arduino\coba_yutup1\coba_yutup1.ino:1:

C:\Users\Amanda Rahmat\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

In file included from C:\Users\Amanda Rahmat\Documents\Arduino\coba_yutup1\coba_yutup1.ino:1:0:

C:\Users\Amanda Rahmat\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

Using library firebase-arduino-master at version 0.1 in folder: C:\Users\Amanda Rahmat\Documents\Arduino\libraries\firebase-arduino-master 
Using library ArduinoJson-6.x at version 6.10.0 in folder: C:\Users\Amanda Rahmat\Documents\Arduino\libraries\ArduinoJson-6.x 
Using library Arduino_JSON at version 0.1.0 in folder: C:\Users\Amanda Rahmat\Documents\Arduino\libraries\Arduino_JSON 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\Amanda Rahmat\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266HTTPClient 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

And my project
#include <Firebase.h>
#include <FirebaseArduino.h>
#include <FirebaseCloudMessaging.h>
#include <FirebaseError.h>
#include <FirebaseHttpClient.h>
#include <FirebaseObject.h>

#include <Arduino_JSON.h>

#include <ESP8266WiFi.h>

  #define wifiName  "ManMan"
  #define  wifiPass  ""
 #define FIREBASE_HOST ""
 #define FIREBASE_AUTH "" //FIREBASE AUTHORIZATION

  int ledpower=5;
void setup() {
  Serial.begin(115200);
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
 
  Serial.print("Connecting to ");
  Serial.println(wifiName);
 
  WiFi.begin(wifiName, wifiPass);
 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());   //You can get IP address assigned to ESP

 Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
 pinMode(ledpower,OUTPUT);
}
 

void loop() {
 int ledStatus = Firebase.getInt("ledStatus");
 Serial.println(ledStatus);
if (ledStatus==1){
 digitalWrite(ledpower,HIGH);
}
else{
   digitalWrite(ledpower,LOW);
 }
}

closed 2019-04-19 by bblanchon