bblanchon/ArduinoJson #957: StaticJsonBuffer

Player76, 2019-04-01:
I have the version 6 of the Json files but I receive this
[image]

smuellener, 2019-04-01:
With ArduinoJson v6 you have to use StaticJsonDocument instead of StaticJsonBuffer. See the following link and video:
https://arduinojson.org/v6/doc/upgrade/
https://www.youtube.com/watch?v=Kn0xtXlAot8

Player76, 2019-04-01:
The issue is in the FirebaseArduino.h file, and I'm not advanced enough to understand his update

bblanchon, 2019-04-01:
Hi @Player76,

Since you cannot upgrade the code, I recommend that you **downgrade the library**.
Open the library manager, find ArduinoJson, select the version 5.13.5 and click install.

Regards,
Benoit

Player76, 2019-04-01:
Thanks it helped

closed 2019-04-03 by bblanchon