bblanchon/ArduinoJson #958: serializeJson to SPIFFS seems not working for me after upgrade 5->6

tschaban, 2019-04-01:
Hi Folks, I have just upgraded from version 5 to 6, and after a couple of modifications required for version 6 I still can't save to SPIFFS using serializeJson 
* Platform: ESP8266 / Testing on ESP12e 

I prepared a sample sketch and I can't spot the issue.  The similar code on AJson 5 works very well. 
Would you advise please? Much appriciate. 

```
#include <ArduinoJson.h>
#include <FS.h>

struct someStructure {
  char helloWorld[11];
};

void load() {
  Serial.println(F("Loading file"));
  File configFile = SPIFFS.open("test.json", "r");

  size_t size = configFile.size();
  std::unique_ptr<char[]> buf(new char[size]);
  configFile.readBytes(buf.get(), size);

  StaticJsonDocument<100> doc;
  DeserializationError error = deserializeJson(doc, configFile);
  if (error) {
    Serial.println(F("Error: deserializeJson"));
    Serial.println(error.c_str());
  }
  Serial.print(F("serializeJson = "));
  serializeJson(doc, Serial);

  configFile.close();
}

void save() {
  Serial.println(F("Saving file"));

  File configFile = SPIFFS.open("test.json", "w");

  if (configFile) {
    Serial.println(F("File opened"));
    StaticJsonDocument<100> doc;
    //  DynamicJsonDocument doc(2048);

    doc["helloWorld"] = "Hello World";
    Serial.print(F("Serializing to file. Size = "));
    uint16 size = serializeJson(doc, configFile);
    Serial.println(size);

  }
  configFile.close();
}

void setup() {

  Serial.begin(9600);

  if (SPIFFS.begin()) {
    Serial.println(F("\nSPIFFS mounted"));
  } else {
    Serial.println(F("\nFailed to mount SPIFFS"));
  }

  Serial.println(F("Formating SPIFFS "));
  if (SPIFFS.format()) {
    Serial.println(F("done"));
  } else {
    Serial.println(F("ERROR"));
  }

  delay(1000);

  save();

  delay(1000);

  load();
}

void loop() {}
```

[image]

bblanchon, 2019-04-03:
Hi @tschaban,

Your program read the file twice.
When you call [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/), the read pointer is already at the end of the file, so it doesn't find any JSON input and returns [IncompleteInput](https://arduinojson.org/v6/api/misc/deserializationerror/) (there is no dedicated error code for an empty input).

Simply remove the call to `readBytes()` and your program will work:

```c++
void load() {
  Serial.println(F("Loading file"));
  File configFile = SPIFFS.open("test.json", "r");
  
  StaticJsonDocument<100> doc;
  DeserializationError error = deserializeJson(doc, configFile);
  if (error) {
    Serial.println(F("Error: deserializeJson"));
    Serial.println(error.c_str());
  }
  Serial.print(F("serializeJson = "));
  serializeJson(doc, Serial);

  configFile.close();
}
```

Regards,
Benoit

tschaban, 2019-04-03:
I see, thanks @bblanchon 
I'll play with it late evening and confirm it. Warm regards.

tschaban, 2019-04-05:
Hi @bblanchon,

I wanted to confirm that it works well. Please close the issue.
Good piece of the code !
thx

closed 2019-04-05 by tschaban

bblanchon, 2019-04-06:
Cool! Thanks for the update @tschaban, and good luck with your project.