bblanchon/ArduinoJson #959: Deserializing a string with escaped quotes to JsonObject worked in v5 but not in v6

smuellener, 2019-04-02:
I would like to deserialize a JsonObject partly to save memory by not deserializing properties of unknown size. 

Example:

I have a string containing a serialized partly known object:

```
{
    "UnknownProperty": "valueWithUnknownSize",
    "KnownProperty": "valueWithKnownSize"
}
```

Is it possible to only deserialize a known specific property from a serialized object?

E.g.:

```
StaticJsonDocument<64> doc;
deserializeJsonSpecificProperty(doc, example, "KnownProperty");
```

This would be helpful for a library that only knows (the required size for deserializing) part of the object.  Thanks!

bblanchon, 2019-04-03:
Hi @smuellener,

I've thought about such a feature for a long time, but I don't see that coming in the near future.

For this feature to be complete, it would need to support nested members, arrays, multiple selectors, etc.
In the end, it would probably require a complete JSONPath implementation.
It's not impossible, but I'm concerned with the impact on code size and performance.
Is it something we want to do on a microcontroller? I'm not sure.

Anyways, I'll keep this issue opened as a feature request, but it will probably stay there for a while.

Meanwhile, you can work around this issue by skipping the part of the input you don't care.
Have a look at [How to deserialize a very large document?](https://arduinojson.org/v6/faq/how-to-deserialize-a-very-large-document/) for explanations.

Regards,
Benoit

closed 2020-02-12 by bblanchon

smuellener, 2020-02-13:
Very nice solution thanks!

bblanchon, 2020-02-13:
`DeserilizationOption::Filter()` is more complicated than the solution you proposed, but it's also more powerful because you can select multiple members.

bblanchon, 2020-03-22:
This feature was included in [ArduinoJson 6.15.0](https://arduinojson.org/news/2020/03/22/version-6-15-0/).

commit def6a81b (2020-02-12): Added `DeserializationOption::Filter` (closes #959)
commit 66b12da4 (2020-02-12): Added `DeserializationOption::Filter` (closes #959)