bblanchon/ArduinoJson #961: Hi Benoit Blanchon, I'd like to discuss with my issue -435 in more details

Nuba123, 2019-04-06:
I upgrade my ArduinoJson Library to V6.10.0 from library manager. And I also check the upgradation tutorial of your's and run the two examples (JsonGenerator+JsonParser). But I didn't get any output in serial monitor for the JsonParser according to your video tutorial. 

So, Is that might be an issue to get error (described in issue number-435) to work with DHT11 sensor wired with ESP8266. Whereas, ESP is connected with my laptop using an USB cable and get connected via COM-3. Following are my included library
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>                                                    
#include <FirebaseArduino.h>                                                
#include <DHT.h> 
#include <Adafruit_Sensor.h> 
                                                        
#define FIREBASE_HOST "yourprojectname.firebaseio.com"                          
#define FIREBASE_AUTH "W5oYCR6eqZDevkPlRR0QULmMiqi"                                              
                                                     
void setup() {
  Serial.begin(921600);
pLease help me to sort it out as soon as possible.

mobizt, 2019-04-07:
Farebase Arduino is now not support ArduinoJson v6 only v5 https://github.com/FirebaseExtended/firebase-arduino/issues/426

bblanchon, 2019-04-08:
Hi @Nuba123,

First, we need to understand why the examples are not working.
As far as I know [JsonParserExample.ino](https://arduinojson.org/v6/example/parser/) and [JsonGeneratorExample.ino](https://arduinojson.org/v6/example/generator/) work on all platform without modification.
Please try again and verify that the baud rate is 9600 in the serial monitor.

Also, as @mobizt mentioned "FirebaseArduino" requires ArduinoJson 5.

Regards,
Benoit

Nuba123, 2019-04-08:
Thanks @bblanchon @mobizt  It helps. I installed ArduinoJson V13.0.1. And uninstalled V-6.10.1. Now the errors are resolved which I mentioned in my post (issue-435). But the new issue is arise. Would you please take a quick look on it and give me a quick feedback?

I got the error message in IDE serial monitor as "DHT11 Sensor Read Failed".

I set the baud rate in serial monitor 115200 same as in sketch Serial.begin(115200);
I connected the DHT11 data pin (2) to ESP8266's D2 (GPIO 4) and define in sketch #define DHTPIN 4 . I also did test in either way. Following is the script.   

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>                                                    
#include <FirebaseArduino.h>                                                
#include <DHT.h> 
//#include <ESP8266HTTPClient.h>
//#include <SoftwareSerial.h>  
#include <Adafruit_Sensor.h> 
                                                        
#define FIREBASE_HOST "yourproject.firebaseio.com"                          
#define FIREBASE_AUTH "000000000000000000000"            
#define WIFI_SSID "MMMM"                                            
#define WIFI_PASSWORD "XXXXXXXXXi"                                    
#define DHTPIN 4                                                           
#define DHTTYPE DHT11                                                       
DHT dht(DHTPIN, DHTTYPE);
                                                     
void setup() {
  Serial.begin(115200);
  delay(1000);                
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);                                     
  Serial.print("Connecting to ");
  Serial.print(WIFI_SSID);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("Connected to ");
  Serial.println(WIFI_SSID);
  Serial.print("IP Address is : ");
  Serial.println(WiFi.localIP());                                            
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);                              
  dht.begin();                                                              
}
void loop() { 
  float h = dht.readHumidity();                                              
  float t = dht.readTemperature();                                           
    
  if (isnan(h) || isnan(t)) {                                                
    Serial.println(F("Failed to read from DHT sensor!"));
    return;
  }
  
  Serial.print("Humidity: ");  Serial.print(h);
  String fireHumid = String(h) + String("%");                                         
  Serial.print("%  Temperature: ");  Serial.print(t);  Serial.println("Â°C ");
  String fireTemp = String(t) + String("Â°C");                                                     
  delay(4000);
  
  Firebase.pushString("nubarealtimedb.firebaseio.com/DHT11/Humidity", fireHumid);                                  
  Firebase.pushString("nubarealtimedb.firebaseio.com/DHT11/Temperature", fireTemp);   
}

bblanchon, 2019-04-23:
Hi @Nuba123,

I'm sorry, this question is not about ArduinoJson anymore.
I cannot afford to answer questions outside of the scope of the library.
Thank you for your understanding.

Regards,
Benoit

closed 2019-04-23 by bblanchon

Nuba123, 2019-04-28:
Hi @bblanchon. No worries, I got my answer on the issues with DHT Sensor and Firebase from Github. And solved the problem.