bblanchon/ArduinoJson #962: parseObject() failed

ziyadov, 2019-04-07:
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>

 
const char* ssid     = "blablabla";
const char* password = "blablabla";
const char* host = "ziyadov.000webhostapp.com"; //replace it with your webhost url
String url;
  int count = 0;
void setup() {
  Serial.begin(115200);
  delay(100);
  pinMode(71, OUTPUT);
  pinMode(D2, OUTPUT);
  pinMode(D3, OUTPUT);
  digitalWrite(D1, 0);
  digitalWrite(D2, 0);
  digitalWrite(D3, 0);
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password); 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("Netmask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway: ");
  Serial.println(WiFi.gatewayIP());
  digitalWrite(D1, 1);
  delay(500);
  digitalWrite(D1, 0);
  delay(500);
}
void loop() {

  Serial.print("connecting to ");
  Serial.println(host);

  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  if (count == 0){
    url = "/api/led/status.php?id=1";
    count = count + 1;
    Serial.println("Here1");
  }
  else if (count == 1){
    url = "/api/led/status.php?id=2";
    count = count + 1;
    Serial.println("Here2");
  }
  else if (count == 2){
    url = "/api/led/status.php?id=3";
    count = count + 1;
    Serial.println("Here3");
  }
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(500);
  String section="header";
  while(client.available()){
    String line = client.readStringUntil('\r');
    //Serial.print(line);
    // we’ll parse the HTML body here
    if (section=="header") { // headers..
     
      if (line=="\n") { // skips the empty space at the beginning 
        section="json";
      }
    }
    else if (section=="json") {  // print the good stuff
      section="ignore";
      String result = line.substring(1);
       
      // Parse JSON
      int size = result.length() + 1;
      char json[size];
      result.toCharArray(json, size);
      StaticJsonBuffer<200> jsonBuffer;
      JsonObject& json_parsed = jsonBuffer.parseObject(json);
      if (!json_parsed.success())
      {
        Serial.println("parseObject() failed");
        return;
      }
      String led = json_parsed["led"][0]["status"];

      if(count == 1){
        if(led == "on"){
          digitalWrite(D7, 1);
          delay(100);
          Serial.println("D1 is On..!");
        }
        else if(led == "off"){
          digitalWrite(D1, 0);
          delay(100);
          Serial.println("D1 is Off..!");
        }
      }
      else if(count == 2){
        if(led == "on"){
          digitalWrite(D2, 1);
          Serial.println("D2 is On..!");
        }
        else if(led == "off"){
          digitalWrite(D2, 0);
          Serial.println("D2 is Off..!");
        }
      }
      else if(count == 3){
        if(led == "on"){
          digitalWrite(D3, 1);
          Serial.println("D3 is On..!");
        }
        else if(led == "off"){
          digitalWrite(D3, 0);
          Serial.println("D3 is Off..!");
        }
        count = 0;
      }

      if (count == 3)
        count = 0;
      

    }
  }
  Serial.println();
  Serial.println("closing connection");
  delay(3000);
}
[image]

bblanchon, 2019-04-10:
Hi @ziyadov,

Please read [Why parsing fails?](https://arduinojson.org/v5/faq/why-parsing-fails/).
Also, consider upgrading to ArduinoJson v6 because it returns an [error code](https://arduinojson.org/v6/api/misc/deserializationerror/) instead of just a boolean.

Regards,
Benoit

closed 2019-04-19 by bblanchon

hwathanh, 2019-04-21:
I have used ArduinoJson v.5.13.5 for this [https://www.instructables.com/id/How-to-Control-LED-Using-ESP8266-NodemCU-Lua-WiFi-/](url) on host xyz.000webhostapp.com, it's ok with my project. But when i change the host to another for example: XYZ.com it's parseObject() failed.
Is this lib written only for host 000webhost.com or i have to config my host something anymore?
Thanks!