bblanchon/ArduinoJson #963: JsonObject' has no member named 'getOrCreateMember'

anctsys, 2019-04-08:
Arduino : 1.8.5 (Linux), Carte : "Arduino MKR GSM 1400"
ArduinoJson 6.10.0

Very very powerfull library 
When i using getOrCreateMember() on json object i have this error message :
error: 'ArduinoJson::JsonObject' has no member named 'getOrCreateMember'

Merci pour ce travail excellent

bblanchon, 2019-04-09:
Hi @anctsys,

It looks like you still have ArduinoJson 5 installed.
I'm pretty sure the error would be `ArduinoJson6100_000::ObjectRef has no member...` with v6.

Make sure the right version of the library is installed.
You can check the version with [ARDUINOJSON_VERSION](https://arduinojson.org/v6/api/misc/version/).

Regards,
Benoit

anctsys, 2019-04-09:
Hi
I check the version "Using ArduinoJson version 6.10.0"
Best regards
anct

bblanchon, 2019-04-10:
Hi @anctsys,

So you're sure that it's version 6.10 but you're still getting this error... hummm.
Please provide an [MCVE](https://stackoverflow.com/help/mcve); you can use [wandox.org](https://wandbox.org/permlink/HBrUTPgJmauExToR) to share the code.

Regards, 
Benoit

closed 2019-04-23 by bblanchon

Aearsis, 2019-05-05:
@anctsys Just came across this - same error, same version. According to a changelog, `getOrCreateMember` was renamed to `getOrAddMember` in version 6.9.

anctsys, 2019-05-11:
Hi @bblanchon , Hi @anctsys 
A lot of thanks
I tested this some of code
#include <ArduinoJson.h>

void setup() {

  // Initialize serial port
  Serial.begin(9600);
  while (!Serial) continue;
  Serial.print("Using ArduinoJson version ");
  Serial.println(ARDUINOJSON_VERSION);
  StaticJsonDocument<200> doc;

  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, json);
  // Test if parsing succeeds.
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }

  // Fetch values.
  //
  // Most of the time, you can rely on the implicit casts.
  // In other case, you can do doc["time"].as<long>();
  const char* sensor = doc["sensor"];
  long time = doc["time"];
  double latitude = doc["data"][0];
  double longitude = doc["data"][1];
  doc.getOrAddMember("coucou");

  // Print values.
  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
  serializeJson(doc, Serial);
}

void loop() {
  // not used in this example
}

And the result is good
Using ArduinoJson version 6.10.0
gps
1351824120
48.756081
2.302038
{"sensor":"gps","time":1351824120,"data":[48.75608,2.302038],"coucou":null}