bblanchon/ArduinoJson #966: Unable to use to my cpp project -- (usin arm compiler in VC++)

carter3rix, 2019-04-11:
Am building a cpp application using VC++ 2008... I have put the ArduinoJson dir in my project 
but anytime i compile... i get the error below.... Help...
`error ArduinoJson requires a C++ compiler, please change file extension to .cc or .cpp `

bblanchon, 2019-04-11:
Are you sure the compiler supports C++?

carter3rix, 2019-04-12:
> Are you sure the compiler supports C++?

Yes it does, am building a POS application with telpo sdk... Any help

bblanchon, 2019-04-15:
I'm not familiar with this toolchain, can you give more details?
I never saw a C++ compiler that doesn't define `__cplusplus`, are you sure it supports C++?
Did you set the file extension to `.cpp` as advised in the error message?

bblanchon, 2019-04-23:
@carter3rix, any update?

closed 2019-05-09 by bblanchon