bblanchon/ArduinoJson #971: "etc does not name a type" error

Archeresque, 2019-04-21:
Hello, I am receiving the does not name a type error on the following very simple lines of code:

```
StaticJsonDocument<200> doc;
doc["type"] = "TEM";
root["serialNumber"] = "234455";
root["uuid"] = "0123456789";

```
The curious thing is that I am receiving this error on my own sketch but I am able to compile the examples no problem. I am using the same include statement from the examples but for whatever reason, my sketch is having an issue. 

I am using the 6.x branch of the library and I am on Arduino 1.8.9 on Mac OS. 

Thank you for any help!

bblanchon, 2019-04-23:
Hi @Archeresque,

It seems that you should replace `root` with `doc`.
If it still doesn't work, please copy the compiler output.

Regards,
Benoit

Archeresque, 2019-04-25:
> Hi @Archeresque,
> 
> It seems that you should replace `root` with `doc`.
> If it still doesn't work, please copy the compiler output.
> 
> Regards,
> Benoit

Hello, my compiler output is below. Apologies about `root` and `doc` I originally had the lines with `root` as comments but I had removed the backslashes erroneously when I pasted the code here so that is not the issue. I may be wrong but the compiler output implies that I'm not referencing the library correctly but the sketches in the examples compile normally with seemingly the exact same include statements and syntax. 

Compiler output:

Arduino: 1.8.9 (Mac OS X), Board: "Adafruit Feather HUZZAH ESP8266, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Authorization:21:1: error: 'doc' does not name a type
 doc["type"] = "TEM";
 ^
Authorization:22:1: error: 'doc' does not name a type
 doc["serialNumber"] = "234455";
 ^
Authorization:23:1: error: 'doc' does not name a type
 doc["uuid"] = "0123456789";
 ^
exit status 1
'doc' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bblanchon, 2019-04-25:
Where is `doc` defined?

Archeresque, 2019-04-25:
It is defined as in my original post as a StaticJsonDocument.

bblanchon, 2019-04-25:
Do the ArduinoJson examples compile?

Archeresque, 2019-04-25:
Yes they do.

bblanchon, 2019-05-09:
Hi @Archeresque,

Did you make any progress?
Did you try to backtrack from the examples to find what made the compilation fail?

Regards,
Benoit

closed 2019-05-16 by bblanchon