bblanchon/ArduinoJson #972: problem of serializeJson in ArduinoJson 6 with ESP8266 WiFiClientSecure

liuzheng1990, 2019-04-21:
Hi, I'm using ArduinoJson to compose the body of HTTPS requests. I've got some problem when trying to serialize the json document to WiFiClientSecure object. Here are some sample codes. I'm using **ArduinoJson 6.10.0** with ESP8266 Arduino core board both **2.4.2** and **2.5.0**. Roughly the same error message appeared. 

```
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <WiFiClientSecure.h>
#include <WiFiClient.h>

#define HOSTNAME <my cloud server's IP>
#define PORT 443
#define FINGER_PRINT <self-signed cert's fingerprint>

WiFiClientSecure http;
// WiFiClient http; // when change to WiFiClient, it compiled.

void setup() {
  // put your setup code here, to run once:
  WiFi.begin(<wifi_ssid>, <wifi_password>);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("WiFi connection established! IP: "); Serial.println(WiFi.localIP());
}

void loop() {
  if (!http.connect(HOSTNAME, PORT))
  {
    Serial.println(F("WiFiManagerError: Failed to connect to the server!"));
    http.stop();
    delay(5000);
    return;
  }

/* need to comment the fingerprint verification if using WiFiClient */
  if (!http.verify(FINGER_PRINT, HOSTNAME)) 
  {
    Serial.println(F("WiFiManagerError: SSL certificate does not match!"));
    http.stop();
    delay(5000);
    return;
  }

  StaticJsonDocument<200> jdoc;
  jdoc["user_id"] = "Alice";
  jdoc["age"] = 26.5;
  
  http.print("POST"); http.print(" "); http.print("/"); http.println(" HTTP/1.1");
  http.print("HOST: "); http.println(HOSTNAME);
  http.println("User-Agent: haiyan");
  jdoc["device_id"] = "nbzheng";
  size_t len = measureJson(jdoc);
  http.println("Content-Type: application/json");
  http.print("Content-Length: "); http.println(len);
  http.println("Connection: close");
  http.println();
  serializeJson(jdoc, http);
}
```
Here is the error message:

> In file included from /home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:10:0,
                 from /home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:34,
                 from /home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/bear/DataBank/codelab/arduino_lab/wifi_tutorial/json_wificlientsecure/json_wificlientsecure.ino:2:
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp: In instantiation of 'void ArduinoJson6100_000::TextFormatter<TWriter>::writeRaw(char) [with TWriter = axTLS::WiFiClientSecure]':
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:58:19:   required from 'void ArduinoJson6100_000::TextFormatter<TWriter>::writeFloat(T) [with T = float; TWriter = axTLS::WiFiClientSecure]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:56:5:   required from 'void ArduinoJson6100_000::JsonSerializer<TWriter>::visitFloat(ArduinoJson6100_000::Float) [with TWriter = axTLS::WiFiClientSecure; ArduinoJson6100_000::Float = float]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:29:51:   required from 'void ArduinoJson6100_000::VariantData::accept(Visitor&) const [with Visitor = ArduinoJson6100_000::JsonSerializer<axTLS::WiFiClientSecure>]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:14:5:   required from 'void ArduinoJson6100_000::variantAccept(const ArduinoJson6100_000::VariantData*, Visitor&) [with Visitor = ArduinoJson6100_000::JsonSerializer<axTLS::WiFiClientSecure>]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Variant/VariantRef.hpp:351:33:   required from 'void ArduinoJson6100_000::VariantConstRef::accept(Visitor&) const [with Visitor = ArduinoJson6100_000::JsonSerializer<axTLS::WiFiClientSecure>]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:21:39:   required from 'void ArduinoJson6100_000::JsonDocument::accept(Visitor&) const [with Visitor = ArduinoJson6100_000::JsonSerializer<axTLS::WiFiClientSecure>]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:21:3:   required from 'typename ArduinoJson6100_000::enable_if<(! ArduinoJson6100_000::IsWriteableString<TPrint>::value), unsigned int>::type ArduinoJson6100_000::serialize(const TSource&, TPrint&) [with TSerializer = ArduinoJson6100_000::JsonSerializer; TSource = ArduinoJson6100_000::StaticJsonDocument<200u>; TPrint = axTLS::WiFiClientSecure; typename ArduinoJson6100_000::enable_if<(! ArduinoJson6100_000::IsWriteableString<TPrint>::value), unsigned int>::type = unsigned int]'
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:102:55:   required from 'size_t ArduinoJson6100_000::serializeJson(const TSource&, TDestination&) [with TSource = ArduinoJson6100_000::StaticJsonDocument<200u>; TDestination = axTLS::WiFiClientSecure; size_t = unsigned int]'
/home/bear/DataBank/codelab/arduino_lab/wifi_tutorial/json_wificlientsecure/json_wificlientsecure.ino:58:27:   required from here
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:136:13: error: no matching function for call to 'axTLS::WiFiClientSecure::write(uint8_t)'
     _length += _writer.write(static_cast<uint8_t>(c));
             ^
/home/bear/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:136:13: note: candidates are:
In file included from /home/bear/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/WiFiClientSecure.h:23:0,
                 from /home/bear/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:42,
                 from /home/bear/DataBank/codelab/arduino_lab/wifi_tutorial/json_wificlientsecure/json_wificlientsecure.ino:1:
/home/bear/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h:46:10: note: virtual size_t axTLS::WiFiClientSecure::write(const uint8_t*, size_t)
   size_t write(const uint8_t *buf, size_t size) override;
          ^
/home/bear/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h:46:10: note:   candidate expects 2 arguments, 1 provided
/home/bear/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h:48:10: note: size_t axTLS::WiFiClientSecure::write(Stream&)
   size_t write(Stream& stream); // Note this is not virtual
          ^
/home/bear/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/WiFiClientSecureAxTLS.h:48:10: note:   no known conversion for argument 1 from 'uint8_t {aka unsigned char}' to 'Stream&'
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Sorry it's a bit long and messy. I think it's a type conversion problem.
When `WiFiClientSecure` is replaced by `WiFiClient` for HTTP POST, not HTTPS, it compiled. In addition, when I used **ArduinoJson 5**, it worked without problems. I do like the `JsonDocument` instead of `JsonBuffer` and still wants to know how to solve this problem in ArduinoJson 6.

Is there anything I missed? Thanks very much for your help!

Best
Zheng

bblanchon, 2019-04-22:
Hi @liuzheng1990,

Please use the following workaround until I find a better solution:

```c++
serializeJson(jdoc, (Client&)http);
```

Thank you very much for reporting this issue.

Regards,
Benoit

liuzheng1990, 2019-04-23:
Hi Benoit,
I'll give it a try this evening. Thanks very much for your help!

Regards,
Zheng

closed 2019-04-23 by bblanchon

bblanchon, 2019-04-23:
Hi Zheng,

[ArduinoJson 6.10.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.10.1) should fix this issue.

Regards,
Benoit

liuzheng1990, 2019-04-23:
The workaround works as well. Interestingly, this type conversion is not needed in `deserializeJson` :) Thanks very much for your help, Benoit!

Regards,
Zheng

commit 12f9aac4 (2019-04-23): Fixed "no matching function for call to write(uint8_t)" (closes #972)