bblanchon/ArduinoJson #975: Compilation error for LOLIN (WEMOS) D1 R1 and R2

Roman1820, 2019-04-28:
While compiling for LOLIN (WEMOS) D1 R1 and R2 I get an error:
Compilation error for LOLIN (WEMOS) D1 R1 as well as 
Compilation error for LOLIN (WEMOS) D1 R2 & mini.
The IDE version is 1.8.9 and the ArduinoJson version is 6.10.1.

bblanchon, 2019-04-29:
Hi @Roman1820,

I got no error here with 1.8.6.
Can you please copy/paste the compiler's output?

Best regards,
Benoit

Roman1820, 2019-04-29:
Hi Benoit,
There is nothing else in the computer's output but 2 orange lines at the bottom:
"Exit status 1
Compilation error for LOLIN (WEMOS) D1 R1".
Best regards,
Roman

Roman1820, 2019-04-29:
Hi, Benoit,
It was my fault. The error was caused by a structure as the second parameter in the function "loadConfiguration" that I tried to transfer by the address. Like "void loadConfiguration (const char *filename, Configuration &Config)".
Thank you a lot for your library!
Best regards,
Roman

bblanchon, 2019-04-30:
You're welcome! Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project 😉

closed 2019-04-30 by bblanchon