bblanchon/ArduinoJson #976: Having issue while updating from V5 to V6 in serializing

thedim-witted, 2019-04-29:
I am using an ESP32 to send data using the json. It had been working fine with the V5, however, there was a time limit issue where it would get stuck at obj.printTo(client) for several minutes. I couldn't understand the problem so I thought about migrating to the V6. After making the changes I don't seem to get the serializePretty value in my serial and so I'm not sure if any data is being pushed to the cloud. 

Code with V5
`DynamicJsonBuffer jsonBuffer(14700);
float X1dataArray[SAMPLE_SIZE], Y1dataArray[SAMPLE_SIZE], Z1dataArray[SAMPLE_SIZE];
float multiplier = ADXL345_MG4G_MULTIPLIER * SENSORS_GRAVITY_STANDARD;

void loop() {
  int iterator;
  connection_iterator = 0;
  connection_iterator2 = 0;

  jsonBuffer.clear();
  JsonObject &rootObject = jsonBuffer.createObject();
  JsonObject &dataObject = rootObject.createNestedObject("data");
  JsonArray &X1dataObject = dataObject.createNestedArray("X1");
  JsonArray &Y1dataObject = dataObject.createNestedArray("Y1");
  JsonArray &Z1dataObject = dataObject.createNestedArray("Z1");

  hspi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE3));
  start_time = micros();
  for (iterator = 0; iterator < SAMPLE_SIZE; iterator++) {
    readAccel(multiplier);
    X1dataArray[iterator] = (ret[0] - cal_x);
    Y1dataArray[iterator] = (ret[1] - cal_y);
    Z1dataArray[iterator] = (ret[2] - cal_z);
    delayMicroseconds(500);
  }
  hspi->endTransaction();

  for (iterator = 0; iterator < SAMPLE_SIZE; iterator++) {
    X1dataObject.add(X1dataArray[iterator]);
    Y1dataObject.add(Y1dataArray[iterator]);
    Z1dataObject.add(Z1dataArray[iterator]);
  }

  rootObject["coreid"] = String(low, HEX) + String(high, HEX);
  //rootObject["coreid_high"] = String(high, HEX);
  rootObject["sample_time"] = int(time_taken);
  rootObject["firmware_version"] = FIRMWARE_UPDATE_VERSION;
  rootObject.prettyPrintTo(Serial);
  Serial.print("connecting to ");
  Serial.println(host);
  while ((!client.connect(host, httpsPort)) && !setup_mode) {
    connection_iterator++;
    Serial.println("connection failed! Retrying..");
    delay(5000);
    if (connection_iterator >= 10)
      ESP.restart();
  }
  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.print("requesting URL: ");
  Serial.println(url);
  client.println(String("POST ") + url + " HTTP/1.0");
  client.println(String("Host: ") + host);
  client.println("Cache-Control: no-cache");
  client.println("Content-Type: application/json");
  client.print("Content-Length: ");
  client.println(rootObject.measureLength());
  client.println();
 // Serial.println("AvadaKedavara");
  rootObject.printTo(client);
  Serial.println("request sent");
  while ((client.connected()) && !setup_mode) {
    connection_iterator2++;

    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
    Serial.println("Waiting for response");
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
    if (connection_iterator2 >= 30) {
      Serial.println("No response from server");
      break;
    }
  }

  String line = client.readStringUntil('\n');
  if (line.startsWith("{")) {
    Serial.println("esp32/Arduino CI successfull!");
  } else {
    Serial.println("esp32/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
  digitalWrite(LED_BUILTIN, LOW);

}`

Code with V6

`DynamicJsonDocument jsonBuffer(14700);
float X1dataArray[SAMPLE_SIZE], Y1dataArray[SAMPLE_SIZE], Z1dataArray[SAMPLE_SIZE];
float multiplier = ADXL345_MG4G_MULTIPLIER * SENSORS_GRAVITY_STANDARD;

void loop() {
  int iterator;
  connection_iterator = 0;
  connection_iterator2 = 0;

  jsonBuffer.clear();
  JsonObject rootObject = jsonBuffer.as<JsonObject>();
  JsonObject dataObject = rootObject.createNestedObject("data");
  JsonArray X1dataObject = dataObject.createNestedArray("X1");
  JsonArray Y1dataObject = dataObject.createNestedArray("Y1");
  JsonArray Z1dataObject = dataObject.createNestedArray("Z1");

 
  hspi->beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE3));
  start_time = micros();
  for (iterator = 0; iterator < SAMPLE_SIZE; iterator++) {
    readAccel(multiplier);
    X1dataArray[iterator] = (ret[0] - cal_x);
    Y1dataArray[iterator] = (ret[1] - cal_y);
    Z1dataArray[iterator] = (ret[2] - cal_z);
    delayMicroseconds(500);
  }
  hspi->endTransaction();

  for (iterator = 0; iterator < SAMPLE_SIZE; iterator++) {
    X1dataObject.add(X1dataArray[iterator]);
    Y1dataObject.add(Y1dataArray[iterator]);
    Z1dataObject.add(Z1dataArray[iterator]);
  }

  rootObject["coreid"] = String(low, HEX) + String(high, HEX);
  //rootObject["coreid_high"] = String(high, HEX);
  rootObject["sample_time"] = int(time_taken);
  rootObject["firmware_version"] = FIRMWARE_UPDATE_VERSION;
  serializeJson(dataObject, Serial);
  Serial.print("connecting to ");
  Serial.println(host);
  
  Serial.print("Heap:  ");
  Serial.println((int)esp_get_free_heap_size());
  
  while ((!client.connect(host, httpsPort)) && !setup_mode) {
    connection_iterator++;
    Serial.println("connection failed! Retrying..");
    delay(5000);
    if (connection_iterator >= 10)
      ESP.restart();
  }
  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.print("requesting URL: ");
  Serial.println(url);
  client.println(String("POST ") + url + " HTTP/1.0");
  client.println(String("Host: ") + host);
  client.println("Cache-Control: no-cache");
  client.println("Content-Type: application/json");
  client.print("Content-Length: ");
  client.println(measureJson(rootObject));
  client.println();
  Serial.println("AvadaKedavara");
  Serial.print("Heap:  ");
  Serial.println((int)esp_get_free_heap_size());
  serializeJson(dataObject, client);
  Serial.println("request sent");
  Serial.print("Heap:  ");
  Serial.println((int)esp_get_free_heap_size());
  while ((client.connected()) && !setup_mode) {
    connection_iterator2++;

    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
    Serial.println("Waiting for response");
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
    if (connection_iterator2 >= 30) {
      Serial.println("No response from server");
      break;
    }
  }

  String line = client.readStringUntil('\n');
  if (line.startsWith("{")) {
    Serial.println("esp32/Arduino CI successfull!");
  } else {
    Serial.println("esp32/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
  digitalWrite(LED_BUILTIN, LOW);

}`

The serial log where it gets stuck with V5 is:

> 12:28:00.584 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
>12:28:00.584 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
>12:28:00.584 -> [V][ssl_client.cpp:276] send_ssl_data(): Writing HTTP request...
>12:47:11.229 -> [E][ssl_client.cpp:33] handle_error(): UNKNOWN ERROR CODE (004E)
>12:47:11.229 -> [E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -78
>12:47:11.229 -> [V][ssl_client.cpp:245] stop_ssl_socket(): Cleaning SSL connection.
>12:47:11.229 -> request sent
>12:47:12.217 -> esp32/Arduino CI has failed
>12:47:12.217 -> reply was:

The serial log with V6 where it is suppose to print the values is just showing up "null"

thedim-witted, 2019-04-29:
And also, in the place of dataObject I actually intend to send the rootObject in which case the whole transaction is failing with the V6 version.

MartinMueller2003, 2019-05-04:
I am having the same issue. Output of serialize is always 'null'

	StaticJsonDocument<40960> jsonBuffer;
	JsonObject Root = jsonBuffer.as<JsonObject>();
	JsonObject ConfigData = Root.createNestedObject(M_CONFIG_JSON_ROOT);
	ConfigData[M_CONFIG_REVISION_NAME] = M_CONFIG_REVISION;
	ConfigData[M_CONFIG_ID] = m_iConfigId;
	m_sRawConfigData.clear();
	serializeJson (jsonBuffer, m_sRawConfigData);
	DEBUGLINE("Start of saved json data");
	DEBUGLINE(m_sRawConfigData.c_str());
	DEBUGLINE("End of saved json data");
-----------Output on console -------------
void c_Config::CommitConfig() Start of saved json data
void c_Config::CommitConfig() null
void c_Config::CommitConfig() End of saved json data

bblanchon, 2019-05-04:
@thedim-witted, what makes you think the problem is in ArduinoJson?

@MartinMueller2003, you must call `jsonBuffer.to<JsonObject>()` instead of `jsonBuffer.as<JsonObject>()`.
See [JsonDocument::to<T>()](https://arduinojson.org/v6/api/jsondocument/to/) and [JsonDocument::as<T>()](https://arduinojson.org/v6/api/jsondocument/as/).
BTW, you don't even need to call it, see [ArduinoJson 6.8.0: more with less!](https://arduinojson.org/2019/02/04/version-6-8-0/)

MartinMueller2003, 2019-05-04:
That did it for me. Two little letters causing tons of investigation. Thank you for the pointer.

thedim-witted, 2019-05-07:
@bblanchon I actually did some research and found out it is not a problem with the ArduinoJson but the WiFiSecureClient library in the ssl connection. Sorry and Thank you.

closed 2019-05-07 by thedim-witted