bblanchon/ArduinoJson #977: Problem with StaticJsonDocument.as<JsonVariant>() when forwarding a template argument as capacity

nobodyinperson, 2019-05-02:
Hi Benoît,

I am writing a [Json-RPC library based on ArduinoJson](https://gitlab.com/tue-umphy/co2mofetten/arduino-libraries/ArduinoJsonRPC) and cannot figure out a compilation error I keep getting.

My goal is to have a classmethod as template that takes one argument, namely the capacity of a `StaticJsonDocument` in the method. When I hard-code the capacity (bad design), everything works. If I use the template argument as capacity, I get `error: expected primary expression...` at a strange place: when I ise `doc.as<JsonVariant>()`. (The library uses `JsonVariant`s due to its flexibility).

# Wandox Minimal Example

https://wandbox.org/permlink/S4u8O3SSMSbzmJZK

```c++
#include <iostream>
#include "ArduinoJson.h"

class MyClass { 
public:
    template<size_t N>
    void method(void) {
        // StaticJsonDocument<100> doc; // hard-coded capacity works
        StaticJsonDocument<N> doc; // <<-- forwareded template argument raises ”error: expeted primary-expression ...”
        JsonVariant var = doc.as<JsonVariant>();
        var["key"] = "value";
        serializeJson(var, std::cout);
    }
};

int main() {
    MyClass myObject;
    myObject.method<100>();
}

```

Output:

```
prog.cc: In member function 'void MyClass::method()':
prog.cc:10:45: error: expected primary-expression before '>' token
         JsonVariant var = doc.as<JsonVariant>();
                                             ^
prog.cc:10:47: error: expected primary-expression before ')' token
         JsonVariant var = doc.as<JsonVariant>();
                                               ^
```

[Source code line in library](https://gitlab.com/tue-umphy/co2mofetten/arduino-libraries/ArduinoJsonRPC/blob/0baf75e3b10cb24359634d591338736859fcd84b/lib/Server.hpp#L230)

I neither understand this error nor do I know how to look for solutions :-S I know I could use a `DynamicJsonDocument` but following the advice in your book I try to avoid the heap as much as possible :-)

Any help would be much appreciated.

Yann

bblanchon, 2019-05-03:
Hi Yan,

This is one of the dark corners of C++: when you call a template method inside a template method, you must use `. template` instead of `.`.
See: https://wandbox.org/permlink/87rkh7pgULdjcAvQ

BTW, in the case studies of [Mastering ArduinoJson](https://arduinojson.org/book/), I demonstrate how to create a memory-efficient JSON-RPC library.

Regards,
Benoit

nobodyinperson, 2019-05-03:
Thanks Benoît, that works! Sometimes, compiler warnings are really misleading and incomprehensible...

I read your book [Mastering ArduinoJson 6](https://arduinojson.org/book/) with pleasure and learned a lot. I also saw your memory-efficient `JsonRPCClient` implementation. However, I needed an easy-to-use `JsonRPCServer` with user-configurable methods as well with the a limitation of a maximum payload size of 72 bytes (XBee API mode).

I wonder, is it possible to **rename** key in a `JsonObject` without memory leaks (i.e. without removing and re-adding the value)?

bblanchon, 2019-05-04:
Thank you very much for purchasing the book.

Renaming a key used to be possible in ArduinoJson 5 but is currently not supported in version 6.
However, there is no technical reason for this limitation, so it should be reasonably easy to support renames.

Please open a new issue and explain the use case, so I can understand why you need that and find the most appropriate syntax.

nobodyinperson, 2019-05-06:
Done! #979

closed 2019-05-06 by nobodyinperson