bblanchon/ArduinoJson #979: Renaming an object key

nobodyinperson, 2019-05-05:
Currently, [it is not possible to rename an object key efficiently](https://github.com/bblanchon/ArduinoJson/issues/977#issuecomment-489309197). However, renaming an object key can be handy at times. 

For example, I am currently implementing a [Json-RPC library](https://gitlab.com/tue-umphy/co2mofetten/arduino-libraries/ArduinoJsonRPC). The library also provides a server class which processes Json-RPC requests (a `JsonVariant`) and calls appropriate callback methods which in turn fill a Json-RPC response (also a `JsonVariant`). Depending on the success of the callback, either the `"result"` member or the `"error"` member must be filled according to [the specification](https://www.jsonrpc.org/specification). If renaming object keys would be possible with ArduinoJson, the callbacks could just care about filling a `JsonVariant` with a result or an error object and the server would take care of choosing the right key name depending on the callback's success. Currently, the callbacks need to be aware of the Json-RPC specification which I consider a rather clumsy style.

This is just my use case of renaming object keys but I am sure other people would also benefit from this feature.

BTW Thank you so much for this awesome library! :-)

Yann

bblanchon, 2019-05-09:
Hi Yann,

I'm not sure that's the right solution to your problem.
It looks like the callbacks should return a `JsonDocument` or a `String`.
Can you post some pseudo code to make sure we are on the same line?

Regards,
Benoit

nobodyinperson, 2019-05-20:
Hey Benoît, 

Sorry for the response delay.

Currently, I have it implemented like this:

```cpp
// callback function
bool callbackFunc(JsonVariantConst request, JsonVariant response) { 
  // on success
  response["result"] = // something based on request["params"]
  // on error
  response["error"] = // (parts of) error object as of Json-RPC specification
}

// register the function to call when a Json-RPC request 
// with method "methodname" comes in
server.registerMethod("methodname",callbackFunc)

// process an incoming Json-RPC message
// 1. check the "method"
// 2. call the appropriate registered callback and let it write to response
// 3. complete the response with additional data (e.g. "id", "error", etc...)
server.process(JsonVariantConst request, JsonVariant response)
```

So the callback function gets to see the **full request** AND needs to fiddle with the **full response**.

I would rather like to implement it like this:

```cpp
bool callbackFunc(JsonVariantConst params, JsonVariant resultOrError) { 
  // on success
  resultOrError ... // set to something based on params
  return true;
  // on error
  resultOrError ... // set to (parts of) error object as of Json-RPC specification
  return false;
}
```

Based on the return value, the `server.process()` method would then set up the appropriate response object (i.e. if successful, use `"result"`, if not use `"error"` key for the output of the callback, for which object key renaming would be needed). The callback function would behave more like a "function" in the general sense: parameters in, result out. The callback function would not need to know the Json-RPC specification in detail.

However, it works the way it is now. It really is not a pressing matter right now. I just thought that being able to (memory-)efficiently rename object keys would make some things easier.

Cheers,

Yann

closed 2025-02-07 by bblanchon