bblanchon/ArduinoJson #980: Function deserializeJson is error

sipt91, 2019-05-06:
#include <ArduinoJson.h>
void setup() {
  Serial.begin(9600);
  char json[] = "11,12,13,14";  // wrong json format
  StaticJsonDocument<200> jsonBuffer;
  DeserializationError err = deserializeJson(jsonBuffer, json);
  Serial.print(F("err="));
  Serial.println(err.c_str());
}

void loop() {
}

=>Result: err=Ok

bblanchon, 2019-05-07:
Hi @sipt91 

Indeed, ArduinoJson isn't a very strict parser and doesn't pretend to be.
I understand that, in theory, this could be a problem, but in practice, it has never been.

Do you have a concrete use case where this is problematic?

Best Regards,
Benoit

sipt91, 2019-05-08:
Hi @bblanchon 

This's my case.

Thank you.

<pre>
<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><font color="#000000">ArduinoJson</font><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>

<font color="#5e6d03">#include</font> <font color="#434f54">&lt;</font><b><font color="#d35400">SoftwareSerial</font></b><font color="#434f54">.</font><font color="#000000">h</font><font color="#434f54">&gt;</font>
<b><font color="#d35400">SoftwareSerial</font></b> <font color="#000000">esp8266</font><font color="#000000">(</font><font color="#000000">9</font><font color="#434f54">,</font> <font color="#000000">10</font><font color="#000000">)</font><font color="#000000">;</font>

<font color="#00979c">void</font> <font color="#000000">config_device</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">setTimeout</font><font color="#000000">(</font><font color="#000000">3000</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34;AT+RST&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">find</font><font color="#000000">(</font><font color="#005c5f">&#34;ready\r\n&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">F</font><font color="#000000">(</font><font color="#005c5f">&#34;AT+CWMODE=2&#34;</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">find</font><font color="#000000">(</font><font color="#005c5f">&#34;OK\r\n&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">F</font><font color="#000000">(</font><font color="#005c5f">&#34;AT+CWSAP=\&#34;WiFi-IoT\&#34;,\&#34;\&#34;,1,0&#34;</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">find</font><font color="#000000">(</font><font color="#005c5f">&#34;OK\r\n&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">F</font><font color="#000000">(</font><font color="#005c5f">&#34;AT+CIPMUX=1&#34;</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">find</font><font color="#000000">(</font><font color="#005c5f">&#34;OK\r\n&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">F</font><font color="#000000">(</font><font color="#005c5f">&#34;AT+CIPSERVER=1,1234&#34;</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">find</font><font color="#000000">(</font><font color="#005c5f">&#34;OK\r\n&#34;</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#000000">F</font><font color="#000000">(</font><font color="#005c5f">&#34;Config Mode&#34;</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>

 &nbsp;<font color="#5e6d03">while</font> <font color="#000000">(</font><font color="#000000">1</font><font color="#000000">)</font>
 &nbsp;<font color="#000000">{</font>
 &nbsp;&nbsp;&nbsp;<font color="#00979c">String</font> <font color="#000000">read_data</font> <font color="#434f54">=</font> <font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">readStringUntil</font><font color="#000000">(</font><font color="#00979c">&#39;\n&#39;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;<font color="#00979c">String</font> <font color="#000000">split_read_data</font> <font color="#434f54">=</font> <font color="#000000">read_data</font><font color="#434f54">.</font><font color="#d35400">substring</font><font color="#000000">(</font><font color="#000000">read_data</font><font color="#434f54">.</font><font color="#d35400">indexOf</font><font color="#000000">(</font><font color="#00979c">&#39;{&#39;</font><font color="#000000">)</font><font color="#434f54">,</font> <font color="#000000">read_data</font><font color="#434f54">.</font><font color="#d35400">lastIndex [... cut ...]
 &nbsp;<font color="#000000">esp8266</font><font color="#434f54">.</font><font color="#d35400">begin</font><font color="#000000">(</font><font color="#000000">9600</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;<font color="#000000">config_device</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>

<font color="#00979c">void</font> <font color="#5e6d03">loop</font><font color="#000000">(</font><font color="#000000">)</font> <font color="#000000">{</font>
<font color="#000000">}</font>

</pre>

bblanchon, 2019-05-09:
Hi @sipt91,

Thank you for posting this example, but I don't see how a false positive from ArduinoJson could be a problem here.
Of course, I could make the parser more strict, but it would inevitably have a cost in term of code size.
Also, there are many other places where the parser is forgiving, so I don't think it's the right direction.

Best Regards,
Benoit

sipt91, 2019-05-10:
Hi @bblanchon,

This is test case. Thank you very much. :)

1) arduino receive string **{"conf":["SSID","PASSWORD"]}** from esp8266.
Result is:
err=OK
ssid_ap=SSID
password_ap=PASSWORD
=> function parse_data return true
2) arduino receive string **SSID,PASSWORD** (wrong json format) from esp8266.
Result is:
err=OK
ssid_ap=null
password_ap=null
=> function parse_data return true (It should be wrong)

bblanchon, 2019-05-10:
Hi @sipt91,

`parse_data()` should check the presence of `conf` anyway.

Take this example: `{}`
It's a valid JSON document, right?
And yet, you'd get:
err=OK
ssid_ap=null
password_ap=null

I'll try to make the parser pickier (if that's cheap), but it won't solve your problem.

Best Regards,
Benoit

sipt91, 2019-05-10:
Hi @bblanchon,

I understand your solution.
Thank you very much. :)

bblanchon, 2019-05-16:
On AVR, this feature adds 64 bytes to the size of the parser; I could not do better 😞

closed 2019-05-16 by bblanchon

sipt91, 2019-05-17:
No problem. Thank you very very much. :)

commit 90c1d549 (2019-05-16): Made `deserializeJson()` more picky about trailing characters (closes #980)