bblanchon/ArduinoJson #982: Compile error: invalid initialization of non-const reference of type ArduinoJson

gerardpalk, 2019-05-07:
I am attempting to read 25 rows of data from a SQLite DB and write the results into a JSON Array, with each row being a JSON Object within the array.  

I am writing this for an ESP32. (LOLIN D32 PRO).  I am using Arduino IDE and v6.10.1 of ArduinoJson library.

Similar issues were raised with early versions of ArduinoJson 6.  However, these issues seemed to be closed with an indication that the issue had been fixed.

The SQL execute function (db_exec() has a callback function which runs on every row of the result set. The callback function I am using is below:

```C++
static int callback(void *data, int argc, char **argv, char **azColName) {
  JsonObject& row = data_array.createNestedObject();
  int i;
  for (i = 0; i < argc; i++) {
     row[azColName[i]] = argv[i] ? argv[i] : "NULL";
  }
  return 0;
}
```

In the main program I have made the following declarations

```C++
const int json_size = JSON_ARRAY_SIZE(25) + 25*JSON_OBJECT_SIZE(8);
StaticJsonDocument<json_size> station_data;
JsonArray data_array = station_data.to<JsonArray>();
```

I get the following error:

> C:\Users\gerar\Documents\Arduino\weather_esp32_main_1\weather_esp32_functions.ino: In function 'int callback(void*, int, char**, char**)':
>
> weather_esp32_functions:117:50: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson6101_000::ObjectRef&}' from an rvalue of type 'ArduinoJson6101_000::ObjectRef'
>
>    JsonObject& row = data_array.createNestedObject();

If I write `JsonObject row = data_array.createNestedObject();`, I get the same object (representing the last row of the returned results) repeated 25 times in the resulting JSON document. 

Any help with this would be greatly appreciated.  I suspect the issue may be related to my inadequate C++ knowledge.

bblanchon, 2019-05-07:
Hi @gerardpalk,

The right syntax in ArduinoJson 6 is the one without references.
Usually, when strings repeat themselves in a JSON document, it's because ArduinoJson stores a pointer instead of storing a copy of the strings.
However, it doesn't seem to be your case, because all the strings are `char*` and not `const char*`.

Please create an MCVE on wandbox.
Start from here: https://wandbox.org/permlink/CXSArzk5Qd1sBx2T

Also, don't forget to increase the capacity of your `JsonDocument` because it's too small to hold a copy of each string.

Regards,
Benoit

gerardpalk, 2019-05-08:
Thanks Beniot for your quick response.  I will have a go at creating a MCVE, but probably will not get to it before the weekend.  Regards  Gerard

gerardpalk, 2019-05-11:
Thank you for the advice re the correct syntax for ArduinoJson 6 not using references.
 
In the process of working on the MCVE I came up with the following solution, which seems to deal with the issue of the repeating sections:

```C++
static int callback(void *data, int argc, char **argv, char **azColName) {
  JsonObject row = data_array.createNestedObject();
  int i;
  for (i = 0; i < argc; i++) {
    char* field = argv[i];
    row[azColName[i]] = field ? field : NULL;
  }
  return 0;
}
```
This seems to work nicely now.  I have also increased the capacity of the JsonDocument. Thanks for your help.

Regards Gerard

bblanchon, 2019-05-12:
Hi Gerard,

Thank you very much for the update.
I finally understood what was wrong in the first sample.

```c++
// first sample
row[azColName[i]] = argv[i] ? argv[i] : "NULL";

// second sample
row[azColName[i]] = field ? field : NULL;
```

Because you passed the string literal `"NULL"` as the third argument of the conditional operator (`?:`), the compiler deduced the type to be `const char*` instead of `char*`.
Because ArduinoJson received a `const char*`, it stored a pointer to the string, not a copy; and that's why you saw the same string repeated.
In the second sample, you pass `NULL` (ie 0) as the third argument, so the type is deduced to `char*`.

BTW, you realize that `field ? field : NULL`  is strictly equivant to `field`, right?
So, in the end you program should look like:

```c++
static int callback(void *data, int argc, char **argv, char **azColName) {
  JsonObject row = data_array.createNestedObject();
  for (int i = 0; i < argc; i++)
    row[azColName[i]] = argv[i];
  return 0;
}
```

Best Regards,
Benoit

closed 2019-05-12 by bblanchon

gerardpalk, 2019-05-12:
Thanks Benoit for the feedback.  I clearly stumbled on that solution, leaving the quotes off "NULL" was accidental!  Clearly I need a better grip on references and pointers in c++.  I purchased your book last night, so I am hoping the second chapter will help in that regard.  The first chapter was nicely informative and easy to read.  Thanks again.  Gerard

bblanchon, 2019-05-13:
You're welcome Gerard.
Thank you very much for purchasing the book.