bblanchon/ArduinoJson #984: Merging serialized json and getting the format right

gednz, 2019-05-08:
Hi

Not a bug, just need help.
I have 3 devices, 2 are sensors and one is a gateway. They are connected by lora radio and can communicate by sending strings of data. The gateway will receive these strings, combine them into a single json message and then upload it to an MQTT server.
I have created a simulator sketch using ArdunioJson to create a json doc for each sensor and then convert it to a string. those strings are then deseralized into a new temp json doc and then added as objects to the main json doc.

My problem is that the format (brackets) are not quite right for what my MQTT server will accept.
My Sketch outputs:
`[{"SN001":[{"t1":374,"t2":382}]},{"SN002":[{"t1":673,"t2":927}]}]`

It needs to look like:
`{"SN001":[{"t1":374,"t2":382}],"SN002":[{"t1":673,"t2":927}]}`

I think this is because the individual sensors had brackets around them so if there was a way to remove the outside {} when they are being merged might take care of that, which may also solve the [] around the outside. I have put in comments below the output I get on the serial from the devices and the combined json

```
#include <ArduinoJson.h>

void setup() {
  // Initialize Serial port
  Serial.begin(115200);
  while (!Serial) continue;

////////////    SENSOR A     /////////////////////////////////////////////

//we have two sensor readings  for sensor a
float ta1 = 374;
float ta2 = 382;
StaticJsonDocument<200> doca;

JsonArray devicea = doca.createNestedArray("SN001"); //SN001 would be sensor A serial number
JsonObject dataa = devicea.createNestedObject();
    dataa["t1"] = ta1;
    dataa["t2"] = ta2;

  String sensorA;
  serializeJson(doca, sensorA);
  Serial.print("sensorA string:   ");
  Serial.println(sensorA);   //  Output:   {"SN001":[{"t1":374,"t2":382}]}
  Serial.println();

/////////////     SENSOR B   (same code as A)   /////////////////////////////////////////

//we have two sensor readings  for sensor b
  float tb1 = 673;
  float tb2 = 927;
  StaticJsonDocument<200> docb;

JsonArray deviceb = docb.createNestedArray("SN002"); //SN002 would be sensor B serial number
JsonObject datab = deviceb.createNestedObject();
    datab["t1"] = tb1;
    datab["t2"] = tb2;

  String sensorB;
  serializeJson(docb, sensorB);
  Serial.print("sensorB string:   ");
  Serial.println(sensorB);   //  Output:   {"SN002":[{"t1":673,"t2":927}]}
  Serial.println();

/////////////     GATEWAY       /////////////////////////////////////

  StaticJsonDocument<1000> doc;
  StaticJsonDocument<200> tmp;

JsonArray data = doc.to<JsonArray>();     

    deserializeJson(tmp, sensorA);
    data.add(tmp.as<JsonObject>());

    deserializeJson(tmp, sensorB);
    data.add(tmp.as<JsonObject>());

  String output;
  serializeJson(doc, output);
  Serial.print("combined string:   ");
  Serial.println(output);   //  Output:   [{"SN001":[{"t1":374,"t2":382}]},{"SN002":[{"t1":673,"t2":927}]}]
  Serial.println();
  Serial.println();
  Serial.println("JsonPretty:   ");
  Serial.println();
  serializeJsonPretty(doc, Serial);
  Serial.println();
  Serial.println();
 
}

void loop() {
  // not used
}
```

bblanchon, 2019-05-09:
Hi @gednz,

Try this function:

```c++
void combine(JsonDocument& dst, const JsonDocument& src) {
    for (auto p : src.as<JsonObject>())
        dst[p.key()] = p.value();
}
```

See demo:
https://wandbox.org/permlink/k2ZyJ503cdBj2Chn

Regards,
Benoit

gednz, 2019-05-09:
Hi Benoit

Thanks very much for the quick response.
It works perfect!

Regards
Ged

bblanchon, 2019-05-10:
You're welcome Ged. Thank you for using ArduinoJson.
Don't forget to add a star to support the project ;-)

closed 2019-05-10 by bblanchon